uuvsimulator/uuv_simulator #4: link::GetRelativeLinearAccel

joaobrittoneto, 2016-11-10:
I'd say the derivative of linear velocity in body frame is requested [here](https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_gazebo_plugins/uuv_gazebo_plugins/src/HydrodynamicModel.cc#L223).

Gazebo `link::GetRelativeLinearAccel()` does not provide that, once it doesn't include the coriolis and centripetal [terms](https://bitbucket.org/osrf/gazebo/commits/c99082dfee89d44ba463597e1d1c3cd17a61b69a#comment-3404427)

musamarcusso, 2016-11-10:
Do you suggest implementing that computation on this package? Or would it be the case to make a pull request to the Gazebo repository?

joaobrittoneto, 2016-11-10:
I think that having these method on gazebo would be the best option. 
But until it go to gazebo's master, may be the case to do the computation in the package.

sebastianscherer, 2016-11-14:
Good catch! I guess GetRelativeLinearAccel does not do what we assumed it would. Would you like to create a pull request?

sebastianscherer, 2016-11-25:
We looked into this problem in a little bit more detail and put a more detailed description of the problem [here](https://raw.githubusercontent.com/wiki/uuvsimulator/uuv_simulator/files/CrossTerms.pdf)

Both #13 and #10 should fix this issue.

joaobrittoneto, 2016-11-25:
Nice description :+1:

closed 2016-11-28 by sebastianscherer