uuvsimulator/uuv_simulator #59: gazebo_gui-2 issue

hughhugh, 2017-05-25:
Hi, I have install the uuv simulator in Ubuntu 16.04 ROS kinetic using VMware following the wiki steps. I input the commands `roslaunch uuv_descriptions empty_underwater_world.launch` and got the following logs:
....
[ INFO] [1495694990.448495374, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1495694990.594566214, 0.128000000]: Physics dynamic reconfigure ready.
VMware: vmw_ioctl_command error Invalid argument.
Aborted (core dumped)
**[gazebo_gui-2] process has died** [pid 43452, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/name/.ros/log/6751db6c-4111-11e7-9c52-000c29919bea/gazebo_gui-2.log].
log file: /home/name/.ros/log/6751db6c-4111-11e7-9c52-000c29919bea/gazebo_gui-2*.log

So how could I get rid of this issue? Thank you.

sebastianscherer, 2017-05-25:
Hi,

running Gazebo within a virtual machine can cause problems due to the 3D graphics.

Have you seen [this entry on Gazebo answers describing the same problem?](http://answers.gazebosim.org/question/13214/virtual-machine-not-launching-gazebo/) Maybe the solutions mentioned there can help you.

Best,
Sebastian

musamarcusso, 2017-05-25:
Hi,

one quick fix is to run the following in your terminal

export LIBGL_ALWAYS_SOFTWARE=1

Gazebo should stop crashing then, but the simulation can run much slower than on native Ubuntu. If I am not mistaken, VMWare should also have some drivers that should be installed as well. 
Could you try this and give us a feedback?

Best,
Musa

musamarcusso, 2017-05-25:
There are some modules of Gazebo that might not run on the virtual machine, see

musamarcusso, 2017-06-05:
Hi, did you manage to solve your issue? If you did, can you send a message saying how you did it?

ayushzenith, 2019-03-10:
I had the same issue and 

`export LIBGL_ALWAYS_SOFTWARE=1`

managed to solve it 

Thanks

closed 2019-03-12 by musamarcusso

dawnofclarity, 2020-03-14:
A heads up for anyone else finding this when attempting to get Gazebo working in a VMWare VM.

This combination allows partial 3d acceleration:
* Windows 7 Host
* VmWare 15.5.2 build-15785246
* Ubuntu 18.04.4 LTS
* Graphics: SVGA3D; build: RELEASE; LLVM;
* Gazebo multi-robot simulator, version 9.0.0
Before running execute 
* export SVGA_VGPU10=0
(This rolls back the driver to use to an earlier version of OpenGL protocol)

This is enough to achieve 60fps on a modest PC.