uuvsimulator/uuv_simulator #73: Can't control ROV using joystick

luym11, 2017-07-12:
Hi! 
I followed all the steps and successfully compiled all the packages. While trying: 
```
roslaunch uuv_descriptions empty_underwater_world.launch 
roslaunch uuv_descriptions upload_rexrov.launch mode:=default x:=0 y:=0 z:=-20 namespace:=rexrov 
roslaunch uuv_control_cascaded_pid joy_velocity.launch uuv_name:=rexrov model_name:=rexrov joy_id:=0 

```
I found that moving the joystick has no effect on the gazebo simulation. 
I checked the joystick output and it's fine. I tried to do this: 
`rostopic echo /rexrov1/cml vel` 
nothing is publishing and I got the warning
`WARNING: no messages received and simulated time is active.`

Could you kindly advice what should I do? Thank you. 

(I attached the rqt_graph in below
[image]

luym11, 2017-07-12:
Add one thing, if I directly publish to eg., geometry_msgs/Twist, the ROV will move as the command.

musamarcusso, 2017-07-13:
Hi, I will have a look at this later today when I get my joysticks to test. Could you run `jstest-gtk` and send me an image of the joysticks you get registered in the list?

luym11, 2017-07-13:
Thanks a lot! I'll try to figure out it today, too. Here's the screenshot.  

[image]

And btw, are you guys doing any project using this uuv_simulator? (I'm trying to do a simulation for a research project now

musamarcusso, 2017-07-14:
Hi @luym11,
regarding the issue with the joystick, I just tested the controlling the ROV with my joystick here (XBox 360 Joystick) and it worked fine. I had a look on the model of the joystick that you are using and it can very well be that the mapping is different than the default mapping used for the XBox 360 being used per default by the teleop node. 
There is a launch file for with a default joystick mapping for the Logitech Extreme 3D Pro in the `uuv_gazebo` package.
You can try running 
`roslaunch uuv_descriptions empty_underwater_world.launch`
and then 
roslaunch uuv_gazebo rexrov_default_logitech_joy.launch joy_id:=0`
Maybe then the mapping will be compatible with you joystick and it will react to it. If it still doesn't work properly, you can check the correct index of the axis on your joystick with `jstest-gtk` and give the correct axis to the rexrov_demo launch file like in the Logitech Pro example in [this launch file](https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_gazebo/launch/rexrov_demos/rexrov_default_logitech_joy.launch).

Regarding your question: yes, this package is being used and developed in the scope of the EU-Project SWARMs (http://swarms.eu/). It has been used as a tool to test algorithms and mission plans for different tasks. 

Let me know if the solution worked and don't hesitate to contact us with any issues/questions.

Best,

Musa

luym11, 2017-07-17:
Hi Musa, 

I bought a new joystick to test and it just arrived XD. 
Yeah, it works with logitech joy launch file now. I guess the problem I previously have is that joy stick doesn't have a compatible configuration. 

Thank you very much. 

Yimeng

closed 2017-07-17 by luym11