uuvsimulator/uuv_simulator #77: <neutrally_buoyant> tag does nothing.

atomoclast, 2017-07-13:
Hello, 

I am trying to implement a vehicle in UUV Sim, and I am trying to just get my vehicle to float, and I even set the `<neutrally_buoyant>1</neutrally_buoyant>` tag as shown, but my vehicle keeps sinking the second it is uploaded. Is there any advice you can give to help get this vehicle floating neutrally or even positively buoyant?

musamarcusso, 2017-07-14:
Hi, 

thanks for reporting, I'll check that. Meanwhile, you can have your vehicle positively buoyant if you set the `neutrally_buoyant` flag to zero and set the volume instead, like in the `rexrov` model https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_descriptions/models/rexrov/urdf/rexrov.gazebo.xacro#L32.
The volume is then used to compute the buoyancy force vector.

atomoclast, 2017-07-14:
Hello,

I tried to set the volume to be quite large. Not as large as the `rexrov`, and it was still sinking. I will try that and get back to you, but I tried to shift my volume up and my mass down to force it to float. I'll get back to you with results in a second though.

atomoclast, 2017-07-14:
So I set my volume to `3.1093918 m3` and my mass to `0.123497`, and my vehicle still sank. I may try to set it to exactly what you have the rexrov set for it's volume, and even set the `<box>` tag to what the `rexrov` had as well as it's hydrodynamic model. So I don't know where to check for an issue.

musamarcusso, 2017-07-17:
Did you set  `<neutrally_buoyant>0</neutrally_buoyant>`?

atomoclast, 2017-07-17:
@musamarcusso I tried setting `<neutrally_buoyant>` to `0` and `1`. No real change.

musamarcusso, 2017-07-17:
Hi @atomoclast, is this the vehicle your want to model?

http://www.videoray.com/images/specsheets/2016/2016_P4STANDARDBASE_FINAL.pdf

I will try to replace these parameters here to reproduce this.

atomoclast, 2017-07-17:
@musamarcusso Yes, this is the vehicle I'm trying to model.

atomoclast, 2017-07-17:
I have a mesh file if you need it.

atomoclast, 2017-07-17:
I just tried to launch a test box with what I understand is the buoyancy plugin and attempted it with both the `neutrally_buoyant` tag set to `0` as well as `1`, and the box still sank both times. It was also printing the `[Wrn] [UnderwaterObjectPlugin.cc:207] Relative angular acceleration is invalid -nan` error over and over. 

I have both files below for you to review. Not sure what the issue is. I added the `.txt` extension so I can upload in the comment.

[box.urdf.txt](https://github.com/uuvsimulator/uuv_simulator/files/1153796/box.urdf.txt)
[upload_box.launch.txt](https://github.com/uuvsimulator/uuv_simulator/files/1153797/upload_box.launch.txt)

sebastianscherer, 2017-07-18:
As soon as you see the error `Relative angular acceleration is invalid` our UnderwaterObjectPlugin will not apply any forces or torques anymore, since this is a clear indicator of the physics engine is broken.

I had a look at your box definition:

`<inertia ixx="0" ixy="0" ixz="0" iyy="0" iyz="0" izz="0" />`

This is the culprit. No rigid body should have [moments of inertia](https://en.wikipedia.org/wiki/Moment_of_inertia) nor a mass of zero. Because of the way Gazebo computes accelerations from forces/torques, this will result in a division by zero.

Let us know if this solves your issue.

atomoclast, 2017-07-18:
@sebastianscherer That makes perfect sense. I just used a basic ellipsoid to estimate my inertia for my vehicle, and it's floating now.

On to figuring out the Thruster plugin! I'm getting some issues there too. 

In the gazebo terminal I see: 
```
[Msg] Thruster #0 initialized
[Msg] 	- Link: videoray_pro4/thruster_0
[Msg] 	- Robot model: videoray_pro4
[Msg] 	- Input command Gazebo topic: /videoray_pro4/thrusters/0/input
[Msg] 	- Thrust output Gazebo topic: /videoray_pro4/thrusters/0/thrust
[Msg] Thruster #1 initialized
[Msg] 	- Link: videoray_pro4/thruster_1
[Msg] 	- Robot model: videoray_pro4
[Msg] 	- Input command Gazebo topic: /videoray_pro4/thrusters/1/input
[Msg] 	- Thrust output Gazebo topic: /videoray_pro4/thrusters/1/thrust
[Msg] Thruster #2 initialized
[Msg] 	- Link: videoray_pro4/thruster_2
[Msg] 	- Robot model: videoray_pro4
[Msg] 	- Input command Gazebo topic: /videoray_pro4/thrusters/2/input
[Msg] 	- Thrust output Gazebo topic: /videoray_pro4/thrusters/2/thrust
[Msg] JointStatePublisher::robotNamespace=videoray_pro4
```
In the thruster terminal, I see: 

```
process[videoray_pro4/thruster_allocator-1]: started with pid [26422]
[INFO] [1500390755.909692, 0.000000]: ThrusterManager::update_rate=50
output_dir= /home/andrew/Development/ underwater_ws/install/share/videoray_description/config/videoray_pro4
ThrusterManager: updating thruster poses
conversion_fcn= proportional
conversion_fcn_params= {'gain': 0.00031}
transform: /videoray_pro4/base_link -> /videoray_pro4/thruster_0
could not get transform from: /videoray_pro4/base_link
to: /videoray_pro4/thruster_0
[]
```

atomoclast, 2017-07-18:
I just ran `rosrun tf view_frames` and this was the output. I see the TF tree show a transform between the base_link and the thruster...
[frames.pdf](https://github.com/uuvsimulator/uuv_simulator/files/1156349/frames.pdf)

musamarcusso, 2017-07-18:
The frames between world and the vehicle's base_link are generated by the `message_to_tf` node, which is part of the `hector_localization` package. The RexROV model's launch file example is this 

https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_descriptions/models/rexrov/launch/upload_rexrov_default.launch#L31

How did you state this in the launch file?

atomoclast, 2017-07-18:
@musamarcusso As per the wiki tutorial, I called out the transform as follows: 

```
<!-- Publish state and tf for in relation to the world frame -->
			<group ns="$(arg namespace)">
				<node name="ground_truth_to_tf" pkg="message_to_tf" type="message_to_tf" output="screen">
					<param name="odometry_topic" value="/$(arg namespace)/pose_gt" />
					<param name="frame_id" value="/$(arg world_frame)" />
					<param name="tf_prefix" value="$(arg namespace)" />
				</node>
			</group>
```

It seems different than the line you called out above. I will try that now.

musamarcusso, 2017-07-18:
Can you try replacing the block you sent me by this 
```xml
<include file="$(find uuv_descriptions)/models/common/launch/message_to_tf.launch">
    <arg name="namespace" value="$(arg namespace)"/>
</include>
```
If this fix works, I will correct the tutorial right away.

atomoclast, 2017-07-18:
I tried to replace my block with your block, but it wouldn't work. Only way it worked was will both blocks called together. 

```VelocityControllerNode: initializing node
Thruster #0 - proportional - thrusters/0/input
transform: /videoray_pro4/base_link -> /videoray_pro4/thruster_1
Thruster #1 - proportional - thrusters/1/input
transform: /videoray_pro4/base_link -> /videoray_pro4/thruster_2
Thruster #2 - proportional - thrusters/2/input
transform: /videoray_pro4/base_link -> /videoray_pro4/thruster_3
could not get transform from: /videoray_pro4/base_link
to: /videoray_pro4/thruster_3
[<uuv_thrusters.models.thruster_proportional.ThrusterProportional object at 0x7efc4bdb8a50>, <uuv_thrusters.models.thruster_proportional.ThrusterProportional object at 0x7efc4bdb8c10>, <uuv_thrusters.models.thruster_proportional.ThrusterProportional object at 0x7efc4bd4c190>]
TAM=
[[ 1.          0.          0.        ]
 [ 0.          0.          0.        ]
 [ 0.          0.99999968  0.99999968]
 [ 0.          0.09999997 -0.09999997]
 [ 0.1         0.15249995  0.15249995]
 [ 0.          0.          0.        ]]
ThrusterManager: ready
ThrusterManager: ready
```
Now I need to see why it can't get the TF from thruster 3.

atomoclast, 2017-07-18:
I'm able to launch my version of the `joy_velocity.launch`, but I'm un able to control the vehicle with my controller.

atomoclast, 2017-07-18:
I've been trying to debug things, and I'm comparing the outputs of the `rqt_graph` visualization of the nodes. These are the two networks I'm seeing. Not sure why it isn't working exactly. 

[image]
[image]

musamarcusso, 2017-07-19:
Hi @atomoclast,

no, it is correct in the tutorial. The example in the tutorial is not a funcional example, that's why I put this [README.md](https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_tutorials/uuv_tutorial_rov_model/mesh/README.md) in the mesh folder, so this `prop_mesh_file` has to be edited by the person making the new vehicle model.
The `joy_velocity` is defined per default for the XBox controller mapping. What is the model of the controller that you have?
Do you have your files somewhere in Github? I could try running it and give you a feedback.

atomoclast, 2017-07-19:
@musamarcusso I have uploaded the repo here: https://github.com/atomoclast/videoray_description/tree/feature/DEEP-291

atomoclast, 2017-07-20:
Hello @musamarcusso, have you had a chance to look through this repo yet?

musamarcusso, 2017-07-21:
Yes, I cloned it. I will give a feedback later today.