uuvsimulator/uuv_simulator #80: Problem while trying to control the robot to some waypoint

luym11, 2017-07-17:
Hi! 

Firstly I have a world, a rov and a correspoding dp_controller running and I want to control the rov to a waypoint through terminal commands. I searched the topic list and the service list, found one service that I guess might help. So I tried 
```

rosservice call /rexrov2/go_to "waypoint:
  point:
    x: 0.0
    y: 20.0
    z: 0.0
  max_forward_speed: 1.0
  heading_offset: 0.0
  use_fixed_heading: false" 
```
But it gave me error message as
`ERROR: service [/rexrov2/go_to] responded with an error: error processing request: global name 'i' is not defined`

Could you please kindly advice how should I solve this? Or could you please suggest a better way to control waypoints through sending commands? (different from the tutorial, in which the program reads waypoints from yaml file)

Thank you! 

Yimeng

luym11, 2017-07-17:
My codes attached, in case needed. 
Thanks! 

[multiagent_simulation.tar.gz](https://github.com/uuvsimulator/uuv_simulator/files/1153857/multiagent_simulation.tar.gz)

sebastianscherer, 2017-07-18:
Thanks for reporting this. I was able to reproduce the issue, the problem seems to be [line 168 in pathgenerator.py.](https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_control/uuv_trajectory_control/src/uuv_trajectory_generator/path_generator/path_generator.py#L168)

@musamarcusso can you take a look?

musamarcusso, 2017-07-18:
Hi @luym11, I just made a pull request that includes a solution to this issue. 
Regarding the waypoints, do you want to simply use a go_to command or do you want to send a list of waypoints?

sebastianscherer, 2017-07-18:
Seems fixed to me. @luym11 please let us know in case the problem persists.

closed 2017-07-18 by sebastianscherer

luym11, 2017-07-18:
Hi @musamarcusso , thank you for the response. What I want to do is to make it possible for a node to send a waypoint command to the robot. If the problem is fixed, I guess by calling go_to service in the node should do the job, right? 
If you have other suggestions about how to do this( sending a waypoint or a list of waypoints), please let me know. 
Thank you!

luym11, 2017-07-18:
@sebastianscherer Thank you for the reply. I'll test in a minute and update the status to you.

musamarcusso, 2017-07-18:
Hi @luym11, I think so, you can use the go_to service to send one waypoint at a time. I still need to implement the service adding waypoints to the list also through service, not only using yaml files.

luym11, 2017-07-18:
@musamarcusso Thank you! I'll start by using go_to service. If I develop something that can add waypoints during my project I'll let you know.

luym11, 2017-07-18:
@sebastianscherer Confirmed. It works fine now. Thank you again!