uuvsimulator/uuv_simulator #96: Fossen Added Mass Matrix Questions

atomoclast, 2017-08-14:
Hello! 

I'm back with more questions! In the tutorials and the REXROV example, I see this Fossen added mass matrix. I was keeping it all Zeros for a long while, but now I'm trying to model this matrix via a simple use of a simple geometric approximation xacro macro, and it's outputting the following added mass matrix: 

`1.87 0 0 0 0 0
0 183.1 0 0 0 0
0 0 183.1 0 0 0
0 0 0 0 0 0
0 0 0 0 45.7
0 0 0 0 0 45.7`

Which was causing errors such as: 

`[Wrn] [UnderwaterObjectPlugin.cc:202] Relative linear acceleration is invalid
-nan`

Which reminded me of the prior issue dealing with some issues of zero values in specific inertial areas, so I tried to add an element to my added mass matrix to make it full rank as follows:

`1.87 0 0 0 0 0
0 183.1 0 0 0 0
0 0 183.1 0 0 0
0 0 0 55 0 0
0 0 0 0 45.7 0
0 0 0 0 0 45.7`

Which resulted in the same error. I would love some guidance and education in this subject if at all possible.

For the mean time, I have set the matrix to all zeroes again, with no real problem. 

Thanks in advance!

musamarcusso, 2017-08-15:
Hi @atomoclast, are you using the model with the changes I sent you? I will then test it here to see if I get the same error.

atomoclast, 2017-08-15:
Hello @musamarcusso, thank you for always replying quickly. I am not using the Videoray Pro4 right now, I was trying to model a custom vehicle. I can try with the Pro4 and see if it changes anything.

atomoclast, 2017-08-15:
I just tested it with the Videoray Pro4 model we were working on and got: 
```
[Msg] HMFossen: Using linear damping for forward speed NULL
gzserver: /uuv_gazebo_plugins/uuv_gazebo_plugins/src/HydrodynamicModel.cc:258: gazebo::HMFossen::HMFossen(sdf::ElementPtr, gazebo::physics::LinkPtr): Assertion `(addedMass.size() == 36)&&("Added-mass coefficients vector must have 36 elements")' failed.
Aborted (core dumped)
[gazebo-2] process has died [pid 18953, exit code 134 
```

lurido, 2017-08-29:
Hi @atomoclast, the values you are using for your added mass seem to be very large. Could you try de-escalating in your added mass matrix your 'yy' and 'zz' linear terms by one order of magnitude, and the angular terms by one or even two orders of magnitude?

jlack1987, 2017-09-26:
Hi @lurido, based on the provided formulas in `common.urdf.xacro` from `uuv_descriptions` I don't think they are very large right? In comparison to the rigid body inertia values yeah they are enormous, but if you are approximating the added inertia matrix for your vehicle from a cylinder the following formula is what is provided,
```

jlack1987, 2017-09-26:
It looks like there's some instability in the `added_mass` term, i'm guessing it has to deal with the way you're differentiating the joint velocities to get the accelerations. I changed the added mass matrix to identity and it took longer but it still eventually went unstable.

musamarcusso, 2017-09-26:
I will take a look into this issue.
Meanwhile, have you tried implementing the model with the parameters presented in this [thesis](https://brage.bibsys.no/xmlui/handle/11250/2402215)?

jlack1987, 2017-09-27:
I have not tried that no. I am able to make things stable now with the identity matrix for added mass matrix, but i'm interested in understanding why sometimes it can go unstable with the parameters we were using. I do see the accelerations that are used in the calculation are gotten from differentiating velocity which isn't ideal and is likely the culprit? Based on the comments in the code it looks like that might be the only course of action though given some problems obtaining accelerations from the Gazebo API though which is unfortunate.

musamarcusso, 2017-09-27:
Yes, unfortunately that's the solution for now. We tried for a long time to understand why the accelerations were so high when being computed in the Gazebo API, but until now we couldn't find another solution. If you compare what outputs from the Gazebo API and the numerical computation of the accelerations, Gazebo delivered values even 10 times higher. This was especially bad in the case of slender bodies like some AUVs. The fact that the accelerations are not used in plugins by other Gazebo models made it also difficult to choose the best way to make it work.
The best solution would have been to integrate added-mass and damping terms to the physics engine and compute the dynamics properly without these workarounds. But this quite is difficult, not only to modify the code, but also because Gazebo has no data structure for 6x6 matrices.
Are you working on the same vehicle model as @atomoclast ? If you want, I can also try to test the model to check what could be leading to the unstable behavior.

jlack1987, 2017-09-27:
Bummer but yeah I understand the difficulties that sometimes come with the gazebo API. Aside from that which is outside your control i'm super impressed with how well `uuv_sim` works and it seems well written!

Yes I'm working with @atomoclast and the same model. Ultimately it's just a Videoray model where we used the added mass xacro macro that calculates the added mass matrix from a cylinder given the required parameters. One thing that I haven't played with is the filter parameter used to calculate the accelerations which will change depending on how fast things are being executed. We are running gazebo at 500Hz where the physics thread is sync'd with our controller thread to simulate a real-time environment, so that might have some effect, not sure though.

MohitGupta007, 2020-05-12:
Heyy,
I am working on a project where we are using this package to simulate an AUV in an underwater environment. Is there any way to have non-linear terms in the mass and damping matrix defined in URDF? For example values depending on the current velocity and angle of the bot.   
Thanks in advance

astronaut71, 2021-09-01:
Hi im working on UUV project and have a model of the fish with two propellers. How to calculate the Fossen Added Mass Matrix for my robot?