uuvsimulator/uuv_simulator #98: Directories being linked to cmake targets

jlack1987, 2017-08-18:
When building uuv sim I get a number of the following warnings,
```
WARNING: Target "uuv_underwater_world" requests linking to directory "/opt/ros/kinetic/include".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "uuv_underwater_world" requests linking to directory "/usr/include".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "uuv_underwater_world" requests linking to directory "/opt/ros/kinetic/include".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "uuv_underwater_world" requests linking to directory "/usr/include".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "uuv_underwater_world_ros_plugin" requests linking to directory "/opt/ros/kinetic/include".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "uuv_underwater_world_ros_plugin" requests linking to directory "/usr/include".  Targets may link only to libraries.  CMake is dropping the item.
```
Looks like there are include directories being linked to cmake targets which isn't allowed and cmake is just ignoring them.

musamarcusso, 2017-08-18:
Hi @jlack1987, could you send the whole output you get when trying to compile?

jlack1987, 2017-08-18:
That's pretty much all there is as far as `uuv sim` cmake warnings go

musamarcusso, 2017-08-18:
Did you follow the installation instructions on the wiki https://github.com/uuvsimulator/uuv_simulator/wiki ?

jlack1987, 2017-08-18:
No didn't follow those instructions, but I know how to build it all. The problem is lines like [this](https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_world_plugins/uuv_world_plugins/CMakeLists.txt#L44). 

You can't link include directories with a cmake target which is why cmake throws a warning on those lines and tells you it's going to ignore them.

musamarcusso, 2017-08-21:
Hi @jlack1987, I made some corrections to the cmake files, could you check if you still get the same output when compiling the workspace?

jlack1987, 2017-08-21:
Yep looks like that did it. Thanks!

closed 2017-08-21 by jlack1987