uuvsimulator/uuv_simulator #102: Problem with creating custom vehicel

ali94wadi, 2017-08-21:
Hello
When attempting to launch the newly created vehicle model I get the following:

UnicodeEncodeError: 'ascii' codec can't encode character u'\u201c' in position 10201: ordinal not in range(128)
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/ros/catkin_ws/src/auv/models/rov_example/robots/rov_example_default.xacro' debug:=0 namespace:=rov_example] returned with code [1]. 

Param xml is <param command="$(find xacro)/xacro --inorder '$(find auv)/models/rov_example/robots/rov_example_$(arg mode).xacro' debug:=$(arg debug) namespace:=$(arg namespace)" name="robot_description"/>
The traceback for the exception was written to the log file

the RexROV runs just fine though.

musamarcusso, 2017-09-20:
HI, sorry to respond so late. 
It looks like you have some hidden character that is messing up with the file. If you are still having this problem, you can look into this https://stackoverflow.com/questions/16346914/python-3-2-unicodeencodeerror-charmap-codec-cant-encode-character-u2013-i
Best,
Musa

musamarcusso, 2017-10-26:
Hi @ali94wadi,
I updated the tutorial on how to create a new vehicle
https://uuvsimulator.github.io/tutorials/create_new_vehicle.html
Let me know if this helps to solve your problem, so that I can close this issue.
Best,
Musa

ali94wadi, 2017-10-27:
Thanks for the update
I had attempted to look at encoding, with no positive results
I shall look at the new tutorials very soon, and hopefully close this issue thread

ali94wadi, 2017-11-22:
I have readapted one of the models available to my needs, and for the time being, I shall close this issue thread

closed 2017-11-22 by ali94wadi