uuvsimulator/uuv_simulator #106: Drag force implementation question

jlack1987, 2017-08-30:
I have a question regarding the implementation of the drag model. Based on the comments and variable names and such it looks to me like a quadratic drag model is implemented, but looking through the code I don't see where the velocity is being squared. The method i'm focusing on is pasted below.

```
void HMFossen::ComputeDampingMatrix(const Eigen::Vector6d& _vel,
                                    Eigen::Matrix6d &_D) const
{
  // From Antonelli 2014: the viscosity of the fluid causes
  // the presence of dissipative drag and lift forces on the
  // body. A common simplification is to consider only linear
  // and quadratic damping terms and group these terms in a
  // matrix Drb

  _D.setZero();

  _D = -1 * this->DLin - _vel[0] * this->DLinForwardSpeed;

  // Nonlinear damping matrix is considered as a diagonal matrix
  for (int i = 0; i < 6; i++)
  {
    _D(i, i) += -1 * this->DNonLin(i, i) * std::fabs(_vel[i]);
  }
}
```

I'm unable to find the `Antonelli 2014` publication that's referred to in the comment(send a link if you can, I would like to read it!). Just looking at what is implemented in this method, this looks like a linear drag model.

If we make a simplifying assumption that all the coefficients of `DLinForwardSpeed` are zero and all the coefficients of `DLin` are also zero, then all we need to focus on is `DNonLin`. I would then expect the overall drag force equation to look something like `-DNonLin(i,i) * _vel[i] * std::fabs(_vel[i])` where you get the square of the velocity term.

The above implementation  looks to me like the force due to drag would increase linearly as the velocity of the body increases. Even if we have the coefficients of `DLin` and `DLinForwardSpeed` to be positive and nonzero, it still looks like the drag force would be increasing linearly with body velocity.

Please let me know if I have understood something wrong or am missing something. Thanks!

lurido, 2017-08-31:
Hi @jlack1987,

The reference you asked for details is the book "Underwater Robots" (2014), from Gianluca Antonelli and published by Springer.

In the situation you described, when DLin and DlinForwardSpeed are both zero, notice that you will have DNonLin multiplied by the absolute value of the velocities vector. Here we are composing only the D matrix, which will be later on multiplied by velocity terms in the dynamic equation, generating then the squared velocity term. For a velocity vector 'V', the final product of "abs(V) * V" assures that the Damping term will also be (as it should) dissipative.

musamarcusso, 2017-08-31:
Hi @jlack1987,

the matrix is built in the ComputeDampingMatrix function but then it is multiplied by the velocity again here

https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_gazebo_plugins/uuv_gazebo_plugins/src/HydrodynamicModel.cc#L340

jlack1987, 2017-08-31:
ah ok now I see! Thanks for the clearification!

closed 2017-08-31 by jlack1987

musamarcusso, 2017-08-31:
Glad that that was cleared out. If you find any other parts of the code that might not look correct, please let us know.