uuvsimulator/uuv_simulator #108: Q: water surface modeling?

tve, 2017-09-15:
Question, is there anything in the uuv simulator that models the water surface, such as waves? (And if not, how do get the nice surface animation in the wind farm video?)

sebastianscherer, 2017-09-15:
Hi, there is currently no physical simulation of the water surface or waves. This would be a great feature to add but we currently do not have the capacity to work on that. If there is anything willing to work on that, please let us know.

The waves you might have seen on videos or screenshots are pure visual effects and implemented using [these shaders ](https://github.com/uuvsimulator/uuv_simulator/tree/master/uuv_descriptions/Media/materials/programs) in the material properties.

tve, 2017-09-15:
Ah, thanks for the reply. I'm working on implementing https://www.gamasutra.com/view/news/237528/Water_interaction_model_for_boats_in_video_games.php as a world plugin (gazebo). One issue I'm going to face (I'm not quite that far yet...) is how to represent the moving waves. Right now I'm using a height map, which gives me non-moving waves. That fine for lots of simulations (even though it's weird), but not everything. Given the server/client separation, I'm trying to find out what to do. It seems like I need to essentially run the same algorithm that produces the waves on both sides. Once for the physics and once for the rendering.

musamarcusso, 2017-09-15:
Yes, that can be done. One way would be to have a shader that implements a JONSWAP or Pierson-Moskowitz model and generates the waves for the visualization and one Gazebo plugin that does a similar thing for the models. 
The issue that I see is that these two modules are separated in Gazebo (if you run the simulation with the shader @sebastianscherer said and pause, the waves will still run). Apart from shaders, I don't know any other ways to show something such as waves in the environment. 
Regarding the interaction with the models, one very simple method I found was this implemented in the Kelpie simulator:

https://www.researchgate.net/publication/259741742_Kelpie_A_ROS-Based_Multi-robot_Simulator_for_Water_Surface_and_Aerial_Vehicles

It seems very easy to implement, but I don't know how good it is. I never tried it out.
The biggest issue I see is synchronizing visual and physical plugin.
If you have any ideas on how to implement this, let us know.  I would be very interested in something like it.

musamarcusso, 2017-09-15:
By the way, to start an empty scenario with the waves, you have to run

```
roslaunch uuv_descriptions ocean_waves.launch
```

tve, 2017-09-17:
Thanks for the comments. I have the physics described in my initial post implemented, but not yet tuned. So far it looks pretty convincing for a very first cut. I think I need to make some more progress on flat water before it makes too much sense to look at waves.

What I have is a world plugin that references the water surface model, and boat models. E.g.:
```
    <plugin name="floating" filename="libfloating.so">
      <water_model>ocean</water_model>
      <boat_models>
        <boat>box</boat>
      </boat_models>
    </plugin>
```
It assumes that the water model is a HeightMap (I want to change to a mesh) and that the boat models are triangle meshes with a single link (for now) representing the hull. It then computes the buoyancy, viscous friction, pressure drag, and slamming forces on all submerged triangles of the mesh and applies them to each triangle.

Does it make sense to try and integrate this plugin into uuv-simulator? At this stage I'm trying to avoid ros and it appears that uuv-simulator can run without ros, but I haven't found much info on that. (I have no ros experience, and very little with gazebo). I've also been using the gazebo default branch (i.e. v9), does uuv-simulator compile with v8? I know there are a good number of method in physics that have changed name between v8 and v9 with the move towards the ignition stuff.

sebastianscherer, 2017-09-20:
That video looks quite cool already!

We tried to separate all plugins so there is a non-ROS part and a ROS-plugin that is derived from the pure Gazebo plugin. We so far did not use the Gazebo plugins on their own and it could take some work to get them running without roslaunch etc.

I think it would make sense to integrate this into uuv_simulator and I would be willing to assist with integration when you have a working prototype.

We are still using Gazebo version 7. I am not sure when we will make the switch to Gazebo 8 or 9, since it has to fit into the schedule of the [SWARMs](http://swarms.eu/) project.

musamarcusso, 2017-09-20:
Just one more thing. If you want to change the waves from the world that you launch with 

```
roslaunch uuv_descriptions ocean_waves.launch
```

you can set different parameters in this materials file here

https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_descriptions/Media/materials/scripts/waves.material#L16

The amplitude is set very low, for instance, but you can put other amplitudes and frequencies to see it working.

tve, 2017-09-20:
Thanks for the feedback, Right now I'm trying to get a simple 3D boat model together. Not having experience with that stuff it's gonna take some time...

musamarcusso, 2017-09-20:
You can use this one if you want

https://github.com/uuvsimulator/uuv_simulator/tree/master/uuv_descriptions/world_models/aurora_static

it is static (only used as a kind of marker), but you can modify it into a robot model. And there is also the model for the Kingfisher that you can get here https://github.com/kf/kingfisher. This last one uses the usv_gazebo_plugins from this repository https://github.com/bsb808/usv_gazebo_plugins.

srmainwaring, 2019-01-04:
Hi, I have a similar interest in modelling buoyancy for surface vessels in Gazebo and this seems to be the place where it's being discussed. I've made some progress using the approach mentioned above and would be happy to share if there is any interest.

I have a working example that supports objects defined using gazebo primitives or meshes that may be composed of one or more links. The wave field is modelled using a planar mesh and there are separate plugins for the visuals and the physics which are kept synchronised using the world simulation time. There is a dependancy on the CGAL library as I use its surface mesh package. 

The first example shows a series of boxes connected by universal joints with buoyancy only (no damping or drag forces).

[image]

The second example shows a number of objects in waves subject to buoyancy and some drag forces. The boats are stock from an online 3D warehouse and I've used Blender to decimate the mesh used for collisions to reduce the vertex count used in the buoyancy calculation. 

[image]

PanosMallioris, 2019-02-24:
Hello i have also tried to made a water surface simulation you can check the video

https://www.youtube.com/watch?v=NHfNfg2vhMc

And the repository 

https://github.com/PanosMallioris/Autonomous-Marine-Exploration-Water-Simulation-ROS-GAZEBO

Mojzaar, 2019-03-01:
Hi, 
I was looking for a compatible simulator with ROS to simulate a wavy ocean. 
I studied the Kelpie's paper but I could not find their package in GitHub. Do you know where can I find that?

samchamberlin, 2020-01-14:
@srmainwaring your model is looking amazing! I'm currently working on building a model of a boat rocking on the ocean surface too. I'm pretty new to Gazebo and have limited experience writing plugins. Would you be willing to share the code/plugin your using to simulate the rocking objects? Any help would be very appreciated!

srmainwaring, 2020-01-14:
@samchamberlin you can find the project here: https://github.com/srmainwaring/asv_wave_sim.

JuliusHendrix, 2020-01-16:
@srmainwaring Hey Rhys, I'm new to ROS and Gazebo, and I'm orienting myself on how to simulate a ship lowering something deep into the ocean. Lucky for me, it seems you have made a very useful package for me to use. Thanks a lot in advance! I'm not quite sure how Gazebo packages work, but do you think it would be possible to somehow combine the wave functionality of your package with the underwater modelling of the UUV package?

srmainwaring, 2020-01-20:
@JuliusHendrix yes I think that would be possible, but it would involve changing the way buoyancy is calculated for the UUV if you want the submersible to respond to waves when on the surface as the method used to calculate buoyancy forces is different in the two packages. If you have any questions about the wave simulation package I'll be happy to answer them on the issues section of its repository. For questions about the UUV packages you'll be best to direct those to the authors.

The OSRF VRX team (https://bitbucket.org/osrf/vrx/src/default/) have also made use of the UUV simulator's wave visuals and incorporated my suggestion for synchronising the visuals in the simulation with the physics. They use a volume base buoyancy calculation like the UUV simulator, rather than the surface force based approach used in mine. To start with it might be easier to try combining the approaches in VRX and UUV and moving on to a more complex buoyancy model later if you need it?

JuliusHendrix, 2020-01-20:
@srmainwaring Thank you, I will look into it!