uuvsimulator/uuv_simulator #120: Possible sign error with added  mass coriolis term

jlack1987, 2017-09-26:
Is it possible that the added mass coriolis term calculated in `HydrodynamicModel.cc` has an incorrect sign? I see the equation in your reference that you're following, and it seems correctly calculated in the `ComputeAddedCoriolisMatrix` function, but then the total coriolis forces are computed via `Eigen::Vector6d cor = -this->Ca * velRel` and are negated. 

My other point of reference can be found [here](https://calhoun.nps.edu/bitstream/handle/10945/45123/McMillen_McGhee_Efficient_Dynamic_1995.pdf?sequence=1). That ref presents the math in a different way, but it resolves to the same thing effectively, and I believe it has the sign on the added mass coriolis term the other way around.

jlack1987, 2017-09-26:
Hmm i'm starting to second guess now..

jlack1987, 2017-09-26:
Ok sorry to bother. I believe the signs are correct!

closed 2017-09-26 by jlack1987

musamarcusso, 2017-09-27:
No worries, I was also checking here to be sure :)