uuvsimulator/uuv_simulator #121: ned_joint rpy value?

kdkalvik, 2017-09-28:
I am trying to make use the thruster plugin for a ROV and I am not sure why the ned_joint in urdf/rov_example_base.xacro from the temple to create a new thruster ... has ${pi} as the value for roll? And is there a separate forum/chat room to ask questions about the plugins?

musamarcusso, 2017-09-28:
Hi @kdkalvik, I was just working on this right now. I merged some changes and added a script that can generate update template files for a new robot description so you don't have to copy all the files from the tutorial. If you get the latest version from the `master` branch, you can go to you `src` folder from the catkin workspace you are working on and run

`rosrun uuv_assistants create_new_robot_model --robot_name <robot_name>`

this will create a catkin package with the data structure needed to model ROVs and AUVs.

You are right, the NED link is actually not needed anymore since it is dealt with within the plugin now. I also corrected that from the RexROV model and will take it out of the `rov_example` template.

The is no chat room, but you can contact me or my colleagues per e-mail, if you want.

kdkalvik, 2017-09-28:
perfect thanks!!

closed 2017-09-28 by kdkalvik