uuvsimulator/uuv_simulator #126: Possible error in buoyancy force implementation

jlack1987, 2017-09-29:
In trying to do dynamics compensation control I think I have found a problem with the buoyancy implementation. For example(model is not a surface vessel) looking first at this block in `BuoyantObject.cc`
```
if (!this->isSurfaceVessel)
  {
    if (z + height / 2 > 0 && z < 0)
    {
      this->isSubmerged = false;
      volume = this->volume * (std::fabs(z) + height / 2) / height;
    }
    else if (z + height / 2 < 0)
    {
      this->isSubmerged = true;
      volume = this->volume;
    }

    if (!this->neutrallyBuoyant || volume != this->volume)
      buoyancyForce = math::Vector3(0, 0, volume * this->fluidDensity * this->g);
    else if (this->neutrallyBuoyant)
      buoyancyForce = math::Vector3(
          0, 0, this->link->GetInertial()->GetMass() * this->g);
  }
```
Where the buoyancy force is correctly computed as `vol*density*g`. This is the buoyancy force in world frame though, not in body frame because gravity is in world frame. This force then gets applied to the body in the same file with the following line,
```
this->link->AddForceAtRelativePosition(buoyancyForce, this->GetCoB());
```
I think the problem is that the buoyancy force that's being applied here is still in world frame and is never rotated to body coordinates before being applied to the body. This results in the force/torque that buoyancy applies to the body never changes even as the robot rotates around.

The way I have fixed it for my testing is to change the `AddForceAtRelativePosition` line to,
```
this->link->AddRelativeForce(pose.rot.GetInverse().RotateVector(buoyancyForce));
this->link->AddRelativeTorque(this->GetCoB().Cross(pose.rot.GetInverse().RotateVector(buoyancyForce))); 
```
I think this results in a correct buoyancy model. Please let me know what you think.

sebastianscherer, 2017-09-29:
Hi,

the naming of these functions in Gazebo is rather unfortunate and [frequently seems to confuse people](http://answers.gazebosim.org/question/16968/confused-with-addlinkforce/) I remember also having similar doubts given the name of this function. But AddForceAtRelativePosition in fact expects the force to be provided wrt a world frame. Only the position at which the force should be applied is provided relative to the link frame.

You can trace this back to the [equivalent function dBodyAddForceAtRelPos in ODE ](http://robotics.naist.jp/~akihiko-y/doxy/ode0.9/ode_8cpp_source.html#l00562) where the name seems to originate.

The Gazebo team improved the documentation in the latest versions as you can see e.g. [here](https://bitbucket.org/osrf/gazebo/src/58fbe39d9e90bcfabc9f4a2ba911ce934b29b66a/gazebo/physics/Link.hh?at=default&fileviewer=file-view-default#Link.hh-213).

jlack1987, 2017-09-29:
Oh I see, thanks for that link. Based on that documentation(pasted below) it looks like the "RelativePosition" is relative to the links center of mass,
```
\param[in] _relPos Position on the link to add the force. This
    /// position must be expressed in the link's coordinates with respect to
    /// the link's center of mass.
```
So i'm guessing this means that uuv_sim expects the center of buoyancy(i.e. the coords in the `<center_of_buoyancy>` tags) then to be relative to the center of mass then, is that correct?

jlack1987, 2017-10-01:
Ok based on my experiments with compensating for the dynamics with control yeah it looks like uuvsim expects the specified center of buoyancy to be w.r.t the body's center of mass. Do you think you could add this into the documentation? I think the intuitive approach(at least to me) is to specify the center of buoyancy w.r.t the body frame, so I think it would be really helpful to clearify in the docs how uuvsim expects that quantity to be expressed.

closed 2017-10-01 by jlack1987