uuvsimulator/uuv_simulator #154: Thruster axis requirement

jlack1987, 2017-11-07:
Is it a requirement that all rotate about the x-axis? Looking at the code I believe that is true, but I don't see anywhere in the docs where it says it's a requirement.

jlack1987, 2017-11-08:
The line that makes me think it's hard coded to be the x-axis is the following,
```
math::Vector3 force(this->thrustForce.,0.,0.);
```
in `ThrusterPlugin.cc`. I think I can make this more generic where it uses the actual axis to come up with what axis the thrust force should be applied. It would look something like,
```
math::Vector3 force(joint_axis*this->thrustForce);
```
Let me know if you're interested in making this work for thrusters that rotate about axes other than the x-axis and I can make a PR to fix it.

musamarcusso, 2017-11-09:
Hi @jlack1987,

sorry for the delay on answering your question. It is something we hard coded mostly to make our work easier. The rotation axis for the thruster could be set as configurable, but I have to check in the Thruster Manager where changes would be necessary in order to make this work, otherwise the computation and use of the thruster allocation matrix will not work as it is now.
I'll look into it and give you a feedback.

jlack1987, 2017-11-09:
Yeah no problem, if I may be able to help, it's possible in for the python stuff it'll be fixable by simply modifying this chunk in `thruster.py`
```
thrust_body = trans.quaternion_matrix(orientation).dot(
                numpy.array([1, 0, 0, 0]).transpose())[0:3]
```
where ```[1,0,0,0]``` would change depending on what axis the thruster rotates about. There are a few ways that info could be gotten, either by adding another(perhaps optional) element to the urdf where you can specify the thruster axis that defaults to ```<axis>1 0 0</axis>``` if it's not specified. For the python you could also have them specify it in the thruster manager yaml file for each thruster. I don't use the python stuff so i'm not 100% familiar with it. 

I can definitely help with the implementation though if it's something you all are interested in. I feel like it would be a nice generalization that shouldn't be too difficult to implement.

musamarcusso, 2017-11-10:
The joint axis can be read from Gazebo's joint structure, that would be not a big issue. For the thruster manager it could be given in the YAML file, but I would think it can be a little better if this information can be read from the URDF file directly using the URDF Python class and the `robot_description` from the parameter server. It is already used in the manipulator kinematics interface, maybe it can be also extended to the thruster manager. My idea is just to leave the thruster manager as easy to configure as possible, there are already many parameters to setup for the simulation.

jlack1987, 2017-11-11:
Coding it up and it looks like we shouldn't use `urdfdom-py` to parse robot_description bc it throws exception if there are any xml tags that are not in the URDF schema which is actually a really common thing. I can use etree. We have added some ```<hydro>``` tags for the hydrodynamics stuff and it barfs on those. I can just use the `lxml` module and parse it out manually.

jlack1987, 2017-11-12:
Went ahead fixed it in a [PR](https://github.com/uuvsimulator/uuv_simulator/pull/159).

pxalcantara, 2020-05-19:
@musamarcusso I think we can close this issue, right?

closed 2021-03-15 by jlack1987