uuvsimulator/uuv_simulator #155: Thruster_id error when trying to reset simulation

swordsgnat, 2017-11-09:
Hi, 

I'm trying to run reinforcement learning with a UUV, having it "learn" how to use its thrusters to follow a path. As much, my code is set up to reset the Gazebo simulation periodically (when the UUV does too poorly). My problem is, even on the very first reset, before anything is really happened, I get an error I don't know how to handle.

I'm calling Gazebo's reset_simulation method in the line `rospy.wait_for_service('/gazebo/reset_simulation') `, and the error it triggers is the following:

```
... logging to /home/nate/.ros/log/fba998ba-c567-11e7-ae9b-080027ea6344/roslaunch-nate-VirtualBox-18775.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

process[rosout-1]: started with pid [18801]
started core service [/rosout]
Traceback (most recent call last):
  File "/home/nate/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/xacro/xacro.py", line 62, in <module>
    xacro.main()
  File "<string>", line 697, in main
  File "<string>", line 627, in eval_self_contained
  File "<string>", line 553, in eval_all
  File "<string>", line 528, in eval_all
  File "<string>", line 483, in eval_text
  File "<string>", line 470, in handle_expr
  File "<string>", line 444, in eval_expr
  File "<string>", line 418, in eval_term
  File "<string>", line 398, in eval_factor
  File "<string>", line 369, in eval_lit
xacro.XacroException: Property wasn't defined: u'thruster_id'
while processing /home/nate/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/uuv_simulator/uuv_descriptions/models/rexrov/launch/upload_rexrov.launch:
while processing /home/nate/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/uuv_simulator/uuv_descriptions/models/rexrov/launch/upload_rexrov_default.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/home/nate/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/xacro/xacro.py '/home/nate/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/uuv_simulator/uuv_descriptions/models/rexrov/robots/rexrov_default.xacro' debug:=0 namespace:=rexrov] returned with code [1]. 

Param xml is <param command="$(find xacro)/xacro.py '$(find uuv_descriptions)/models/rexrov/robots/rexrov_$(arg mode).xacro' debug:=$(arg debug) namespace:=$(arg namespace)" name="robot_description"/>
The traceback for the exception was written to the log file
```

It seems to be some sort of initialization issue, but for the life of me I can't figure out what it is or how to change it. Does anyone have any ideas?

musamarcusso, 2017-11-09:
Hi @swordsgnat,

actually I had the exact same problems reseting the simulation on my end. I actually at some point gave up on trying to do it. What I did instead was making a simulation wrapper that starts the simulation and then ends it completely at each iteration (I was running some optimization methods where the cost function used the simulation to compute the vehicle's dynamics).

The wrapper is used here https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_evaluation/scripts/smac_wrapper.py#L92 and the source for the wrapper is here https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_evaluation/src/parameter_opt/simulation_runner.py.

One place where I used this was in this use-case 

https://github.com/uuvsimulator/uuv_simulator/tree/master/uuv_smac_opt/uuv_smac_opt_nmb_sm_no_disturbances

Since it was such a specific functionality, I had not planned to include a tutorial in the next few weeks, but I can include a simple example in the [documentation] (https://uuvsimulator.github.io/) to show how it works, if this is what you looking for. 

Let me know if you want to use this wrapper and I start working on this tutorial for the simulation wrapper.

swordsgnat, 2017-11-09:
Hi @musamarcusso, 

Yeah, that sounds like it would be very helpful - if it wouldn't be too much trouble a tutorial on it would be awesome (I'm still pretty new to all this) - thank you kindly for offering!

swordsgnat, 2017-11-15:
No rush at all, but were you gonna put up a tutorial on this?

musamarcusso, 2017-11-15:
Yes, I am working on it. There are some other things I'm also working on in parallel here, that's why it is taking a little while, but I will push the documentation with this tutorial until Friday.

musamarcusso, 2017-11-20:
Hi @swordsgnat,

sorry for the delay. I merged the code and launch files for the example showing the wrapper's functionality. You can check them here 

https://github.com/uuvsimulator/uuv_simulator/tree/master/uuv_tutorials/uuv_tutorial_simulation_runner

and you can try it out by running 

```
rosrun uuv_tutorial_simulation_runner run_grid_search
```

I'll work on the tutorial page in the documentation now to explain everything. Meanwhile you can check the comments on the files of this tutorial's package. If you find any bugs or errors, please let me know.

Best,

Musa

swordsgnat, 2017-11-20:
No problem! Thank you!

closed 2017-11-20 by swordsgnat