uuvsimulator/uuv_simulator #157: Not working with lunar

patrickelectric, 2017-11-11:
The install need to be update to work with lunar.
```sh
roslaunch uuv_descriptions empty_underwater_world.launch                           
[empty_underwater_world.launch] is neither a launch file in package [uuv_descriptions] nor is [uuv_descriptions] a launch file name
The traceback for the exception was written to the log file
```

swordsgnat, 2017-11-13:
I mean, the documentation only has support listed for Indigo and Kinetic, and even ROS recommends using Kinetic in general... is there some reason you have to use Lunar?

musamarcusso, 2017-11-13:
Hi @patrickelectric,

yes, we still haven't done anything specific for lunar, mostly because the project we are in still uses mostly Kinetic, but this error looks more like paths were not updated after the catkin workspace was built. Did you source the setup.sh file in the devel folder before trying to run this launch file?

musamarcusso, 2017-11-16:
Hi @patrickelectric, 

could it be that your catkin workspace is trying to source the `install` folder? If you have no problem not sourcing the `install` folder, you can try running the following 

```
cd ~/catkin_ws
catkin config --no-install
catkin clean --all
```

and then rebuild the workspace

```
catkin build
source devel/setup.sh
```

patrickelectric, 2017-11-16:
ty @musamarcusso, now I have a different error..
[Link.](http://answers.gazebosim.org/question/17841/gre-erroritemidentityexception-resource-with-the-name-console-already-exists/)

musamarcusso, 2017-11-17:
I think the link you sent is broken. Could you copy the error message here?

patrickelectric, 2017-11-17:
Sorry, the link still need approval of admin.
```
auto-starting new master
process[master]: started with pid [13575]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to bd173442-cb28-11e7-9b11-847bebfceff7
process[rosout-1]: started with pid [13588]
started core service [/rosout]
process[gazebo-2]: started with pid [13591]
process[gazebo_gui-3]: started with pid [13593]
process[publish_world_models-4]: started with pid [13601]
Gazebo multi-robot simulator, version 8.1.1
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[ INFO] [1510876155.413449198]: Finished loading Gazebo ROS API Plugin.
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.2.1
[ INFO] [1510876155.426629675]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[Err] [server_main.cc:54] Ogre Error:ItemIdentityException: Resource with the name Console already exists. in ResourceManager::add at /build/ogre/src/ogre/OgreMain/src/OgreResourceManager.cpp (line 156)
[gazebo_gui-3] process has died [pid 13593, exit code 255, cmd /mnt/hd/ros_ws/lunar/install_isolated/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/patrick/.ros/log/bd173442-cb28-11e7-9b11-847bebfceff7/gazebo_gui-3.log].
log file: /home/patrick/.ros/log/bd173442-cb28-11e7-9b11-847bebfceff7/gazebo_gui-3*.log
Start publishing vehicle footprints to RViz
^C[publish_world_models-4] killing on exit
[gazebo-2] killing on exit
/gazebo/get_model_state service is unavailable
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
```

musamarcusso, 2017-12-21:
Hi @patrickelectric,

I don't know if you are still having issues, but I updated the configuration for Travis CI in #176 to build the simulator with ROS Lunar, after adding two packages that are still not available for Lunar, it is at least compiling:

```
cd ~/catkin_ws/src
git clone https://github.com/tu-darmstadt-ros-pkg/hector_localization
git clone https://github.com/ros-teleop/teleop_tools
```

musamarcusso, 2019-03-13:
Now released for `lunar`.

closed 2019-03-13 by musamarcusso