uuvsimulator/uuv_simulator #166: Gazebo 7.9 and OpenCV

duwke, 2017-12-14:
The Getting Started instructions "Using UUV Simulator with ROS Kinetic and Gazebo 7" seem to have a couple of issues, and we have narrowed them down to problems with Gazebo 7.9 and OpenCV.  First, to get uuv_simulator to build, we had to make the changes as described here:

https://github.com/ros-perception/vision_opencv/issues/193
https://github.com/opencv/opencv/pull/10299

We updated the opencv make file as described above, and it did build to completion.  We assume your project is waiting on that same pull request.  

I have catkin installed, so I built with "catkin build," but when trying to launch the samples, I got the following:

> $ roslaunch uuv_descriptions empty_underwater_world.launch
> [empty_underwater_world.launch] is neither a launch file in package [uuv_descriptions] nor is [uuv_descriptions] a launch file name
> The traceback for the exception was written to the log file

I tried the following, with no change:
https://uuvsimulator.github.io/faq.html#after-running-catkin-build-i-still-can-t-start-any-nodes-or-launch-files.  I did notice there was no install directory.
Instead, I used the catkin_make install, and gazebo did launch.

Unfortunately, it seems some models fail to load:
roslaunch uuv_descriptions ocean_waves.launch
[image]
roslaunch uuv_descriptions empty_underwater_world.launch
[image]
Similar screen shots in 7.8.1 (recompiled from source)
[image]
[image]

I am unable to get the RexROV to launch.

Let me know if you would like any log files.

musamarcusso, 2017-12-14:
Hi @duwke,
I actually had similar things happening yesterday while helping a colleague install the simulation. I am not at the office at the moment, but I will have a closer look into it as soon as I can.

musamarcusso, 2017-12-14:
Short question: did you source the `devel/setup.bash` after compilation? Log files would also be great to understand the problem :)

duwke, 2017-12-15:
What logs would you like?  I built 7.8.1, so I'll need to get back to 7.9

musamarcusso, 2017-12-18:
I am still looking into the OpenCV issue, but if you could send log files on failing to launch RexROV that would be great. I made a pull request #167 to avoid having the build crash because of the dependency of the camera plugins with OpenCV, but I am still looking into the issue with Gazebo as well.

duwke, 2018-01-04:
RexRov model looks fine, just the water looks incorrect.  
I used the following commands:
roslaunch uuv_descriptions ocean_waves.launch
roslaunch uuv_descriptions upload_rexrov.launch x:=0 y:=0 z:=-20 mode:=default namespace:=rexrov

Here are my log files
[master.log](https://github.com/uuvsimulator/uuv_simulator/files/1604884/master.log)
[publish_world_models-4.log](https://github.com/uuvsimulator/uuv_simulator/files/1604885/publish_world_models-4.log)
[rexrov-urdf_spawner-1.log](https://github.com/uuvsimulator/uuv_simulator/files/1604886/rexrov-urdf_spawner-1.log)
[roslaunch-duwke-21067.log](https://github.com/uuvsimulator/uuv_simulator/files/1604887/roslaunch-duwke-21067.log)
[roslaunch-duwke-21605.log](https://github.com/uuvsimulator/uuv_simulator/files/1604888/roslaunch-duwke-21605.log)
[rosout.log](https://github.com/uuvsimulator/uuv_simulator/files/1604889/rosout.log)