uuvsimulator/uuv_simulator #180: Failing to build on Kinetic

clydemcqueen, 2018-01-05:
Hi, I'm running into a build problem. On a clean build (Xenial, Gazebo 7, Kinetic), I'm getting the following error:

```
clyde@fastr:~/uuv_catkin_ws$ catkin_make
Base path: /home/clyde/uuv_catkin_ws
Source space: /home/clyde/uuv_catkin_ws/src
Build space: /home/clyde/uuv_catkin_ws/build
Devel space: /home/clyde/uuv_catkin_ws/devel
Install space: /home/clyde/uuv_catkin_ws/install
####
#### Running command: "cmake /home/clyde/uuv_catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/clyde/uuv_catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/clyde/uuv_catkin_ws/install -G Unix Makefiles" in "/home/clyde/uuv_catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/clyde/uuv_catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/clyde/uuv_catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/clyde/uuv_catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/clyde/uuv_catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.8
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 41 packages in topological order:
-- ~~  - oberon4 (metapackage)
-- ~~  - oberon4_control
-- ~~  - oberon4_description
-- ~~  - oberon7
-- ~~  - oberon7_control
-- ~~  - oberon7_description
-- ~~  - uuv_assistants
-- ~~  - uuv_control_cascaded_pid
-- ~~  - uuv_control_msgs
-- ~~  - uuv_control_utils
-- ~~  - uuv_descriptions
-- ~~  - uuv_evaluation
-- ~~  - uuv_gazebo
-- ~~  - uuv_gazebo_plugins
-- ~~  - uuv_gazebo_ros_plugins_msgs
-- ~~  - uuv_manipulators_commons
-- ~~  - uuv_manipulators_description
-- ~~  - uuv_manipulators_msgs
-- ~~  - uuv_sensor_plugins
-- ~~  - uuv_sensor_plugins_ros_msgs
-- ~~  - uuv_simulator
-- ~~  - uuv_smac_opt (metapackage)
-- ~~  - uuv_smac_opt_nmb_sm_no_disturbances
-- ~~  - uuv_smac_opt_nmb_sm_timed_disturbances
-- ~~  - uuv_smac_opt_pid_no_disturbances
-- ~~  - uuv_smac_opt_pid_timed_disturbances
-- ~~  - uuv_teleop
-- ~~  - uuv_thruster_manager
-- ~~  - uuv_tutorial_disturbances
-- ~~  - uuv_tutorial_dp_controller
-- ~~  - uuv_tutorial_seabed_world
-- ~~  - uuv_tutorial_simulation_runner
-- ~~  - uuv_tutorials
-- ~~  - uuv_world_plugins
-- ~~  - uuv_world_ros_plugins_msgs
-- ~~  - uuv_gazebo_ros_plugins
-- ~~  - uuv_manipulators_kinematics
-- ~~  - uuv_manipulators_control
-- ~~  - uuv_sensor_plugins_ros
-- ~~  - uuv_trajectory_control
-- ~~  - uuv_world_ros_plugins
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

(snip)

-- Could not find the required component 'uuv_world_ros_plugins_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "uuv_world_ros_plugins_msgs" with any of the following names:

    uuv_world_ros_plugins_msgsConfig.cmake
    uuv_world_ros_plugins_msgs-config.cmake

  Add the installation prefix of "uuv_world_ros_plugins_msgs" to
  CMAKE_PREFIX_PATH or set "uuv_world_ros_plugins_msgs_DIR" to a directory
  containing one of the above files.  If "uuv_world_ros_plugins_msgs"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  uuv_simulator/uuv_control/uuv_control_utils/CMakeLists.txt:4 (find_package)

```

Thanks.

musamarcusso, 2018-01-10:
HI @clydemcqueen, thanks for the heads up,
thanks to the pull request submitted in #179 this should resolved. On my side I didn't see this error and the Travis builds have been fine. Could you give a small feedback on this building issues?

clydemcqueen, 2018-01-10:
Thanks for the quick response, @musamarcusso -- 

It seems to come down to a difference in build tools. I was using `catkin_make` and your Travis builds are using `catkin build`. I had erroneously assumed that both tools implement the same package.xml spec, but per [this document](http://catkin-tools.readthedocs.io/en/latest/migration.html#implications-of-isolation), packages built with `catkin build` "do not need to define target dependencies on ROS messages built in other packages." I confirmed that `catkin build` and `catkin_make_isolated` build just fine.

I'm using `catkin_build` because it works better with CLion, but perhaps I'm an outlier. Feel free to close this bug.

musamarcusso, 2018-01-11:
Hi @clydemcqueen,
I missed that on the error message you sent. Yes, `catkin build` and `catkin_make` can give different outputs, I had similar issues when I first started using `catkin build`. Thanks for the feedback!

closed 2018-01-11 by musamarcusso