uuvsimulator/uuv_simulator #181: Added Mass

joaobrittoneto, 2018-01-08:
Hi,

I was working in the added mass issue in our project, using [Rock-Gazebo](https://github.com/rock-gazebo) integration, and I came up
with a different approach for determining the added mass effect in an UUV.

I determine a so called `Compensated Effort (C)`, i.e. a 6x6 matrix ( as a function of the rigid body and the added mass matrices) that multiplies the total efforts applied in the UUV, in such a way the resultant velocity's derivatives in Gazebo is the one provided by Fossen's equation.

If someone is interested, here is the paper of the [solution](http://ieeexplore.ieee.org/document/8232205/).

jlack1987, 2018-02-09:
I'm interested in seeing if this improves the simulation behavior as the current implementation has stability issues. I've had to disable it for my underwater sims as the sim goes unstable shortly after spawning.

musamarcusso, 2018-02-09:
Hi @joaobrittoneto, the issue I see with this in the case of the UUV simulator is considering that there are no contact forces. I would have to make some exceptions to make this possible since some of the scenarios we simulate here have manipulators. 

@jlack1987, could you send some more information on the stability issues you are facing?

jlack1987, 2018-02-09:
Unfortunately @musamarcusso it's been awhile since I've played around with it, but I believe the accelerations would go to `nan` very quickly after spawning when using added mass inertia matrices calculated from primitive shapes(ellipsoids I think?). Unfortunately that's about all I can recall.

musamarcusso, 2018-02-09:
Ah ok, I will have a look into it. I haven't modelled vehicles using the ellipsoid for a long time. I will make some tests here. Most of the vehicles I have added have had their parameters identified through experiments.

joaobrittoneto, 2018-02-19:
Hi @musamarcusso, there is a [beforePhysicsUpdate](http://osrf-distributions.s3.amazonaws.com/gazebo/api/9.0.0/classgazebo_1_1event_1_1Events.html#a18e51f6dcd556597aa4fcd35aaa4b8b2) event and a [contact](http://osrf-distributions.s3.amazonaws.com/gazebo/api/9.0.0/classgazebo_1_1physics_1_1Contact.html) class that may be used for the case of contacts. I didn't really try it, but I think it's worth an investigation.

minderx, 2018-06-02:
how to add the damping coef and added mass for DARPA SUBOFF submarine model?
 i know some coeff  are listed on page  27 of http://www.dtic.mil/dtic/tr/fulltext/u2/a227715.pdf but im not completely sure what they are. for example based on the coeff listed on the paper how can i add those to fit the UUV in what order?

<added_mass>
             ??
            </added_mass>

<linear_damping_forward_speed>
              ??
            </linear_damping_forward_speed>