uuvsimulator/uuv_simulator #189: CMake problem on last commit

cosmer-admin, 2018-01-24:
Hi,

I am running a Ubuntu 14.04 LTS with ROS indigo and Gazebo 7.

I have just tried to install the last commit of uuv-s imulator. 
I had an error in the Cmake configuration for this version which desappear if I checkout the 7b6d063 commit : 
The package uuv_world_ros_plugins_msgs was not found in uuv_control CMakeList.

I followed the installation process and I set the following PATH variable

source /opt/ros/indigo/setup.bash
source $HOME/catkin_ws/devel/setup.bash

export GAZEBO_PREFIX=$HOME/catkin_ws/install
export GAZEBO_RESOURCE_PATH=${GAZEBO_PREFIX}/share/gazebo-7.0:${GAZEBO_RESOURCE_PATH}
export GAZEBO_MODEL_PATH=${GAZEBO_PREFIX}/share/gazebo-7.0/models:${GAZEBO_MODEL_PATH}
export GAZEBO_PLUGIN_PATH=${GAZEBO_PREFIX}/lib:${GAZEBO_PREFIX}/lib/x86_64-linux-gnu:${GAZEBO_PLUGIN_PATH}

May be I did something wrong during the install process?

musamarcusso, 2018-01-24:
Hi @cosmer-admin,

I made some changes to make the build compatible with both `catkin_make` and `catkin build` in this pull request  #191 , but I have ROS Kinetic here. Could you check if it works for you and give a quick feedback?

cosmer-admin, 2018-02-07:
Hi,
It seems that it works fine now on both kinetic and indigo. 
Thank you !

musamarcusso, 2018-02-09:
Hi @cosmer-admin, thanks for the feedback! If you find any other bugs or problems, don't hesitate to create a new issue :) I will close this one now.

closed 2018-02-09 by musamarcusso