uuvsimulator/uuv_simulator #195: Issue With Controller Tutorial - Matrix Problem?

swordsgnat, 2018-01-31:
Hey, I've been working with uuv_simulator for a few months now on a virtual machine. I recently got a dedicated Ubuntu machine, and did a fresh install. When I tried to run the custom dynamic positioning controller tutorial as a test, though, I keep running into this weird error, shown below. Anyone have any idea what it is? 

```
gzserver: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:109: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = double; int Size = 6; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
[rexrov/urdf_spawner-4] process has finished cleanly
log file: /home/nate/.ros/log/543ca190-06c5-11e8-8c94-f8b156a60794/rexrov-urdf_spawner-4*.log
Aborted (core dumped)
[gazebo-2] process has died [pid 26500, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode worlds/ocean_waves.world __name:=gazebo __log:=/home/nate/.ros/log/543ca190-06c5-11e8-8c94-f8b156a60794/gazebo-2.log].
log file: /home/nate/.ros/log/543ca190-06c5-11e8-8c94-f8b156a60794/gazebo-2*.log
```

musamarcusso, 2018-02-01:
Hi @swordsgnat, thanks for reporting this. I looked into it now and apparently other projects are having the same issue with Eigen. I will try and see if I can find a solution for this.

musamarcusso, 2018-02-01:
Could you send the log file that was generated after this happened? I can't reproduce it on my PC or on the Travis docker image.

swordsgnat, 2018-02-01:
- I may be missing something, but it seems like the log isn't being generated. The log file it references in the error doesn't exist, though the folder does. I did a fresh install to make sure I hadn't skipped any steps or committed any silly errors, and I got the same thing. Eigen error, no log generated for the crash. 

- Also, on the empty_underwater_world and ocean_waves launches, the water appears yellow or black, like there's some visual issues - not sure if that's directly related or not. Picture below:

[image]

- I ran the custom dynamic positioning controller tutorial a few times to see if it failed the same way every time and if I could see any other errors. It fails on one of two errors each time, either the Eigen one or the below:

```
[ INFO] [1517498096.663029394]: compiled against OGRE version 1.9.0 (Ghadamon)
spawn_model script started
Starting trajectory and waypoint marker publisher
[INFO] [1517498097.019688, 0.000000]: Loading model xml from ros parameter
[INFO] [1517498097.026092, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
Start publishing vehicle footprints to RViz
Tutorial - DP Controller
2018-02-01 10:14:57,383 | INFO | dp_controller_base | Setting controller as model-based
X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  154 (DRI2)
  Minor opcode of failed request:  3 (DRI2CreateDrawable)
  Resource id in failed request:  0x4e00002
  Serial number of failed request:  31
  Current serial number in output stream:  33
[ INFO] [1517498098.442790944]: Stereo is NOT SUPPORTED
[ INFO] [1517498098.442954799]: OpenGl version: 3 (GLSL 1.3).
[gazebo_gui-3] process has died [pid 16981, exit code 1, cmd /opt/ros/kinetic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/nate/.ros/log/a74692cc-0762-11e8-8c94-f8b156a60794/gazebo_gui-3.log].
log file: /home/nate/.ros/log/a74692cc-0762-11e8-8c94-f8b156a60794/gazebo_gui-3*.log
```

in addition, when I control + C to end the program after the crash for the Eigen error, it drops a whole block of text with what looks like a failed assert in the middle of it:
```

Traceback (most recent call last):
  File "/home/nate/catkin_ws/src/uuv_simulator/uuv_tutorials/uuv_tutorial_dp_controller/scripts/tutorial_dp_controller.py", line 161, in <module>
    node = TutorialDPController()
  File "/home/nate/catkin_ws/src/uuv_simulator/uuv_tutorials/uuv_tutorial_dp_controller/scripts/tutorial_dp_controller.py", line 50, in __init__
    super(TutorialDPController, self).__init__(self)
  File "/home/nate/catkin_ws/src/uuv_simulator/uuv_control/uuv_trajectory_control/src/uuv_control_interfaces/dp_controller_base.py", line 75, in __init__
    stamped_pose_only=self._use_stamped_poses_only)
  File "/home/nate/catkin_ws/src/uuv_simulator/uuv_control/uuv_trajectory_control/src/uuv_control_interfaces/dp_controller_local_planner.py", line 80, in __init__
    raise rospy.Exception('No transform found between world and the '
AttributeError: 'module' object has no attribute 'Exception'
No transform found between base_link and base_link_ned for vehicle /rexrov/
ROS shutdown request
base_link transform NED to ENU=
None
[INFO] [1517498384.895625, 0.002000]: ThrusterManager::update_rate=50
Thruster #0 - proportional - thrusters/0/input
Thruster #1 - proportional - thrusters/1/input
Thruster #2 - proportional - thrusters/2/input
Thruster #3 - proportional - thrusters/3/input
Thruster #4 - proportional - thrusters/4/input
Thruster #5 - proportional - thrusters/5/input
Thruster #6 - proportional - thrusters/6/input
Thruster #7 - proportional - thrusters/7/input
Thruster allocation matrix provided!
TAM=
[[ 0.15974269  0.15974269 -0.15974243 -0.15974243  0.70710686  0.70710686
  -0.70710654 -0.70710654]
 [-0.21361012  0.21361012 -0.21360977  0.21360977  0.7071067  -0.7071067
   0.70710702 -0.70710702]
 [ 0.9637702   0.9637702   0.96377032  0.96377032  0.          0.          0.
   0.        ]
 [ 0.43523203 -0.43523203  0.43523188 -0.43523188 -0.09121676  0.09121676
  -0.09121681  0.09121681]
 [ 0.9430935   0.9430935  -0.94309347 -0.94309347  0.09121679  0.09121679
  -0.09121674 -0.09121674]
 [ 0.13688862 -0.13688862 -0.1368884   0.1368884  -0.64879876  0.64879876
   0.64879873 -0.64879873]]
TAM provided, skipping...
ThrusterManager: ready
Invalid output directory for the TAM matrix, dir= None
ThrusterManager: ready
Leaving ThrusterAllocatorNode
caught exception
exiting
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

None of the errors seem to generate the logs that they reference in the crash statement (there are no logs in the long-character-string folder that say "gazebo" at all, actually, just ros, rviz, rexrov, master and such) and I can't see any more errors in the readouts above the crash. 

Hopefully that's a helpful amount of information, sorry.

musamarcusso, 2018-02-02:
Hi @swordsgnat,

I just reinstalled Ubuntu on a new PC here to see if I could reproduce some of these errors. The setup is the following:

* Ubuntu 16.04.3 LTS
* ROS Kinetic
* Gazebo 7.0 (it came with the installation of `ros-kinetic-desktop-full`)

I will test later on with Gazebo 7.8 to see how it goes.

I then installed the following packages (from the instructions in https://uuvsimulator.github.io/installation.html):

```
sudo apt-get install ros-kinetic-gazebo-msgs ros-kinetic-gazebo-plugins ros-kinetic-gazebo-ros ros-kinetic-gazebo-ros-control ros-kinetic-gazebo-ros-pkgs ros-kinetic-effort-controllers ros-kinetic-image-pipeline ros-kinetic-image-common ros-kinetic-perception ros-kinetic-perception-pcl ros-kinetic-robot-state-publisher ros-kinetic-ros-base ros-kinetic-viz python-wstool python-catkin-tools python-catkin-lint ros-kinetic-hector-localization ros-kinetic-joy ros-kinetic-joy-teleop libopencv-dev protobuf-compiler protobuf-c-compiler
```

* Regarding the weird visuals in Gazebo: did you install the proprietary drivers for your graphics card? On my PC I have a NVidia GeForce GTX 1060 6GB and I had to install the driver 384.111. I don't know if that could affect the visuals, but you could try checking that.

* About the tutorial: sorry about that. I recently added a new inertial reference frame `world_ned` that is created with the worlds in `uuv_descriptions` but the tutorial was starting the world without the launch files in `uuv_descriptions`. If you get the lastest version from the `master` branch, that should be fixed.

* I didn't get the Eigen errors with the new install. I still have one PC that needs updating here so I will have a new opportunity to check if these errors with Eigen come up.

I ran the tutorial as follows

```
roslaunch uuv_tutorial_dp_controller start_tutorial_dp_controller_demo.launch
```

swordsgnat, 2018-02-05:
I did a clean install, and now about half the time I run the tutorial I can get it to work. The other half, it still crashes with the same error, shown below:
```
Registered HydrodynamicModel type spheroid
Registered HydrodynamicModel type box
[Msg] Subscribing to current velocity topic: hydrodynamics/current_velocity
[Msg] Name of the BASE_LINK: rexrov/base_link
gzserver: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:109: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = double; int Size = 6; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
Property::loadValue() TODO: error handling - unexpected QVariant type 0.
Aborted (core dumped)
[gazebo-2] process has died [pid 23933, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver --verbose -e ode worlds/ocean_waves.world __name:=gazebo __log:=/home/nate/.ros/log/19e9b67c-0a94-11e8-8ba9-f8b156a60794/gazebo-2.log].
log file: /home/nate/.ros/log/19e9b67c-0a94-11e8-8ba9-f8b156a60794/gazebo-2*.log
```
When it runs, I still get the black-and-yellow artifacting. I double and triple checked that my video drivers were updated - does anyone have any idea what else this might be?

swordsgnat, 2018-02-07:
In attempting to solve the black/yellow ground problem today, I reduced the simulation to a base case (ocean_waves.launch, which runs without throwing any errors) and starting doing tests. I have ocean_waves working properly on one of my virtual machines, with files from a while ago. I started moving over files one by one, recompiling, and checking to see if it solved the error. I can say with certainty that nothing in uuv_descriptions and nothing in the ~/.bashrc file are causing the change. It could be a different file, or the Gazebo update, which I will start testing next.

swordsgnat, 2018-02-08:
I've isolated the "yellow ground" error. On one of my virtual machines, I was able to trigger the error by only by updating Gazebo from 7.8.1 to 7.9.0 (by uninstalling the old Gazebo and then running the line `sudo apt-get install ros-kinetic-gazebo-msgs ros-kinetic-gazebo-plugins ros-kinetic-gazebo-ros ros-kinetic-gazebo-ros-control ros-kinetic-gazebo-ros-pkgs ros-kinetic-effort-controllers ros-kinetic-image-pipeline ros-kinetic-image-common ros-kinetic-perception ros-kinetic-perception-pcl ros-kinetic-robot-state-publisher ros-kinetic-ros-base ros-kinetic-viz python-wstool python-catkin-tools python-catkin-lint ros-kinetic-hector-localization ros-kinetic-joy ros-kinetic-joy-teleop libopencv-dev protobuf-compiler protobuf-c-compiler` as per the tutorial). So, the problem seems to be something in the Gazebo update (or, I guess, potentially an update with one of these associated packages) that's causing the display issue.

swordsgnat, 2018-02-14:
Gazebo has now updated the 7.X line to 7.11, but after a full recompile it seems to not have effected either error (the visual issues or the Eigen one)

swordsgnat, 2018-02-23:
Hey, not to quintuple-post, but I was wondering if anyone else was running into this issue or working on a fix on the Eigen bug or visual effect? I'm a bit of a dead end myself.

musamarcusso, 2018-02-26:
Hi @swordsgnat, have you tried to run the simulation in a docker container?

swordsgnat, 2018-02-27:
I'm running a 32 bit version of Ubuntu - I may be wrong, since I haven't used Docker before, but I think it needs 64 bit to work? Does uuv_simulator require 64 bit as well - could that be a possible discrepancy?

musamarcusso, 2018-02-28:
Maybe, I imagine that most of the libraries are mostly available for 64 bits systems. But I never tested it in a 32 bit machine, I can't be sure of that.

musamarcusso, 2018-02-28:
It might be a problem also in the future, since the support for [the 32 bit Ubuntu is already being dropped](https://itsfoss.com/ubuntu-drops-32-bit-desktop/)

swordsgnat, 2018-02-28:
I can look at switching which Ubuntu I'm using, but I don't think that's the issue - at least, not entirely.  I went and made a new 64-bit VirtualBox virtual machine and loaded everything onto it from scratch. When it was running Gazebo 7.0.0  , running the Custom Controller Tutorial yielded the "Z-Ordering Bug" that was present in early 7.X Gazebo versions:
`gzserver: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreRenderSystem.cpp:546: virtual void Ogre::RenderSystem::setDepthBufferFor(Ogre::RenderTarget*): Assertion 'bAttached && "A new DepthBuffer for a RenderTarget was created, but after creation" "it says it's incompatible with that RT"' failed.`
When I update Gazebo to the latest in the 7.X line (7.11.0) and update the packages as per the tutorial, I run into the same yellow-and-black striped ground error that I referenced before. Interestingly though, this time I'm not triggering the Eigen error. So it seems like that may be solved?

musamarcusso, 2018-03-02:
Could you send me a list of the packages you are installing? Maybe I can try it out here and see if the problem happens as well.

musamarcusso, 2018-03-02:
Update on the textures problem. I just updated another PC with Gazebo 7.9 and the Ogre log file (`$/.gazebo.ogre.log`) shows that Ogre can't find the textures.

```
14:43:32: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource water_water_0076_03_s.jpg in resource group General or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
14:43:32: Error loading texture water_water_0076_03_s.jpg. Texture layer will be blank. Loading the texture failed with the following exception: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource water_water_0076_03_s.jpg in resource group General or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
14:43:32: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource wave_normals.dds in resource group General or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
14:43:32: Error loading texture wave_normals.dds. Texture layer will be blank. Loading the texture failed with the following exception: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource wave_normals.dds in resource group General or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
14:43:32: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource clouds_rt.jpg in resource group General or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
14:43:32: Error loading texture clouds.jpg. Texture layer will be blank. Loading the texture failed with the following exception: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource clouds_rt.jpg in resource group General or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
14:43:32: Added resource location '/home/mam0box/catkin_ws/src/uuv_simulator/uuv_descriptions/world_models/sand_heightmap/meshes' of type 'FileSystem' to resource group 'General' with recursive option
```

I will follow this log file to find the problem.

swordsgnat, 2018-03-06:
The latest change mostly fixed the texturing problem! The central ground texture square in the custom controller tutorial is now showing up dark, and I'm getting some odd crashes every second time I run the tutorial, but I can deal with that!

Thank you very much!

closed 2018-03-06 by swordsgnat

NickSadjoli, 2018-10-26:
Hello, I would like to have this issue reopened if possible, since I am having a similar problem myself. I am currently using Gazebo 7.14, with ROS Kinetic Kame, Ubuntu 16.04 LTS. Initially I ran the ocean_waves.launch and get the following rendering from one of the cameras

[image]

In addition to that, the empty_underwater_world is also giving a very weird texture like below.
[image]

The problem with this is that the rendering error is being carried over into future or other projects that is being built on top of the uuv_simulator as well. Is there a possible missing model or rendering error that I should be able to catch?

Also of note is that this issue doesn't seem to come up when I upgrade to Gazebo 9. However, since some of the projects that I am planning to use or modify was built with Gazebo 7, I would prefer if I can stay at this version if possible.

The ogre logfile is attached here, in case you need a reference.
[ogre.log](https://github.com/uuvsimulator/uuv_simulator/files/2519631/ogre.log)

Thanks and looking forward to any kind of help from y'all!