uuvsimulator/uuv_simulator #198: [Please help me] Error with receive the waypoint and apply to rexrov

phamhoanganhbk, 2018-02-09:
Hi everyone,

I am trying to run 
roslaunch uuv_tutorial_dp_controller start_tutorial_dp_controller_demo.launch
then 
roslaunch uuv_control_utils send_waypoints_file.launch uuv_name:=rexrov

So i have some errors as following:
--------------------------------------------------------------------------
Send a waypoint file, namespace= /rexrov/
Negative start time, setting it to 0.0
Check point #2
Traceback (most recent call last):
  File "/home/pha/catkin_ws/src/uuv_simulator/uuv_control/uuv_control_utils/scripts/send_waypoint_file.py", line 64, in <module>
    String(filename))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 495, in call
    service_uri = self._get_service_uri(request)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 463, in _get_service_uri
    raise ServiceException("service [%s] unavailable"%self.resolved_name)
rospy.service.ServiceException: service [/rexrov/init_waypoints_from_file] unavailable
[rexrov/send_waypoint_file-1] process has died [pid 4701, exit code 1, cmd /home/pha/catkin_ws/src/uuv_simulator/uuv_control/uuv_control_utils/scripts/send_waypoint_file.py __name:=send_waypoint_file __log:=/home/pha/.ros/log/faebb1f8-0d95-11e8-b26f-5891cf6e920e/rexrov-send_waypoint_file-1.log].
log file: /home/pha/.ros/log/faebb1f8-0d95-11e8-b26f-5891cf6e920e/rexrov-send_waypoint_file-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
------------------------------------------
I used the "check_point" and I saw that  there is error at file send_waypoint_file.py (line 48)
    try:
        rospy.wait_for_service('init_waypoints_from_file', timeout=2)
        print 'Check point #1'
    except rospy.ROSException:
        rospy.ROSException('Service not available! Closing node...')
        print 'Check point #2'

Could you give me an idea to fix it, please?
Thank you very much,

musamarcusso, 2018-02-09:
Hi @phamhoanganhbk,

I had seen this issue on my end once, it seems that the node providing this service wasn't loaded before the call for the send waypoint file command. 
If you update the simulator to the latest version in the master branch, I increased the timeout 

https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_control/uuv_control_utils/scripts/send_waypoint_file.py#L49

Check if this fixes it for you.

Best,
Musa

phamhoanganhbk, 2018-02-09:
Hi Musa,

Thank you very much for your answers (it is so fast :D)
I have just updated the latest version in the master branch. But it isn't working too. 
So, I increased the timeout = 60. It is the same errors.

I think maybe i have a problem with ROS.

Have a nice day,
Hoang-anh

musamarcusso, 2018-02-09:
Hi @phamhoanganhbk, you could try calling the service using the command line with `rosservice call` to see if it works.

```
rosservice call /rexrov/init_waypoints_from_file "start_time:
  data:
    secs: 0
    nsecs: 0
start_now: true
filename:
  data: '<path to the waypoint file>'" 
```

musamarcusso, 2018-02-09:
You could also run `rosservice list` and see if the service is available.
Do you have the log file? Maybe it is easier to see there what happened.

phamhoanganhbk, 2018-02-09:
Hi @musamarcusso ,

Here, my log file

> [rospy.client][INFO] 2018-02-09 15:32:25,704: init_node, name[/rexrov/send_waypoint_file], pid[11693]
[xmlrpc][INFO] 2018-02-09 15:32:25,704: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2018-02-09 15:32:25,704: Started XML-RPC server [http://phacosmer:35344/]
[rospy.init][INFO] 2018-02-09 15:32:25,704: ROS Slave URI: [http://phacosmer:35344/]
[rospy.impl.masterslave][INFO] 2018-02-09 15:32:25,705: _ready: http://phacosmer:35344/
[xmlrpc][INFO] 2018-02-09 15:32:25,705: xml rpc node: starting XML-RPC server
[rospy.registration][INFO] 2018-02-09 15:32:25,705: Registering with master node http://localhost:11311
[rospy.init][INFO] 2018-02-09 15:32:25,805: registered with master
[rospy.rosout][INFO] 2018-02-09 15:32:25,805: initializing /rosout core topic
[rospy.rosout][INFO] 2018-02-09 15:32:25,807: connected to core topic /rosout
[rospy.simtime][INFO] 2018-02-09 15:32:25,808: initializing /clock core topic
[rospy.simtime][INFO] 2018-02-09 15:32:25,810: connected to core topic /clock
[rospy.internal][INFO] 2018-02-09 15:32:25,820: topic[/clock] adding connection to [http://phacosmer:42466/], count 0
[rospy.internal][INFO] 2018-02-09 15:32:26,045: topic[/rosout] adding connection to [/rosout], count 0
[rospy.service][ERROR] 2018-02-09 15:33:26,092: [/rexrov/init_waypoints_from_file]: lookup service failed with message [no provider]
[rospy.core][INFO] 2018-02-09 15:33:26,092: signal_shutdown [atexit]
[rospy.internal][INFO] 2018-02-09 15:33:26,094: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2018-02-09 15:33:26,094: topic[/clock] removing connection to http://phacosmer:42466/
[rospy.impl.masterslave][INFO] 2018-02-09 15:33:26,094: atexit

When i tried to run 
`rosservice call /rexrov/init_waypoints_from_file`

> ERROR: Service [/rexrov/init_waypoints_from_file] is not available.

Thank for your help,

phamhoanganhbk, 2018-02-09:
Hi @musamarcusso ,

I saw it seems the service init_waypoints_from_file is not running. 
i found it is declared in file dp_controller_local_planner.py. 
To active this service, do i need to do anything else?  

Thanks for your helps,
Hoang-anh

musamarcusso, 2018-02-09:
No, it should start without any extra command needed. Could you copy here the output you get when you run

```
 roslaunch uuv_tutorial_dp_controller start_tutorial_dp_controller_demo.launch
```

(without trying to call the service for the waypoints)

phamhoanganhbk, 2018-02-09:
Here is the output

> pha@phacosmer:~/catkin_ws$ roslaunch uuv_tutorial_dp_controller start_tutorial_dp_controller_demo.launch
... logging to /home/pha/.ros/log/87db58a2-0db4-11e8-b5ba-5891cf6e920e/roslaunch-phacosmer-22772.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://phacosmer:37727/

SUMMARY
========

PARAMETERS
 * /publish_world_models/meshes/east/plane: [2000, 0.1, 100]
 * /publish_world_models/meshes/east/pose/position: [0, 1000, -50]
 * /publish_world_models/meshes/heightmap/mesh: package://uuv_des...
 * /publish_world_models/meshes/heightmap/model: sand_heightmap
 * /publish_world_models/meshes/north/plane: [0.1, 2000, 100]
 * /publish_world_models/meshes/north/pose/position: [1000, 0, -50]
 * /publish_world_models/meshes/seafloor/plane: [2000, 2000, 0.1]
 * /publish_world_models/meshes/seafloor/pose/position: [0, 0, -100]
 * /publish_world_models/meshes/south/plane: [0.1, 2000, 100]
 * /publish_world_models/meshes/south/pose/position: [-1000, 0, -50]
 * /publish_world_models/meshes/west/plane: [2000, 0.1, 100]
 * /publish_world_models/meshes/west/pose/position: [0, -1000, -50]
 * /rexrov/ground_truth_to_tf_rexrov/child_frame_id: /rexrov/base_link
 * /rexrov/ground_truth_to_tf_rexrov/footprint_frame_id: /rexrov/base_foot...
 * /rexrov/ground_truth_to_tf_rexrov/frame_id: world
 * /rexrov/ground_truth_to_tf_rexrov/odometry_topic: /rexrov/pose_gt
 * /rexrov/ground_truth_to_tf_rexrov/stabilized_frame_id: /rexrov/base_stab...
 * /rexrov/robot_description: <?xml version="1....
 * /rexrov/robot_state_publisher/publish_frequency: 5
 * /rexrov/robot_state_publisher/robot_description: /rexrov/robot_des...
 * /rexrov/thruster_allocator/tam: [[0.1597426884844...
 * /rexrov/thruster_manager/base_link: base_link
 * /rexrov/thruster_manager/conversion_fcn: proportional
 * /rexrov/thruster_manager/conversion_fcn_params/gain: 0.00031
 * /rexrov/thruster_manager/max_thrust: 2000.0
 * /rexrov/thruster_manager/tf_prefix: rexrov
 * /rexrov/thruster_manager/thruster_frame_base: thruster_
 * /rexrov/thruster_manager/thruster_topic_prefix: thrusters/
 * /rexrov/thruster_manager/thruster_topic_suffix: /input
 * /rexrov/thruster_manager/timeout: -1
 * /rexrov/thruster_manager/update_rate: 50
 * /rexrov/tutorial_dp_controller/Kd: [9077.459, 9077.4...
 * /rexrov/tutorial_dp_controller/Ki: [321.417, 321.417...
 * /rexrov/tutorial_dp_controller/Kp: [11993.888, 11993...
 * /rexrov/tutorial_dp_controller/saturation: 5000
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /use_sim_time: True

NODES
  /rexrov/
    ground_truth_to_tf_rexrov (message_to_tf/message_to_tf)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    thruster_allocator (uuv_thruster_manager/thruster_allocator.py)
    trajectory_marker_publisher (uuv_control_utils/trajectory_marker_publisher.py)
    tutorial_dp_controller (uuv_tutorial_dp_controller/tutorial_dp_controller.py)
    urdf_spawner (uuv_descriptions/spawn_model.py)
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    publish_world_models (uuv_assistants/publish_world_models.py)
    rviz (rviz/rviz)
    world_ned_frame_publisher (tf2_ros/static_transform_publisher)

auto-starting new master
process[master]: started with pid [22787]
[... 257 lines cut ...]
Thruster #5 - proportional - thrusters/5/input
Thruster #6 - proportional - thrusters/6/input
Thruster #7 - proportional - thrusters/7/input
Thruster allocation matrix provided!
TAM=
[[ 0.15974269  0.15974269 -0.15974243 -0.15974243  0.70710686  0.70710686
  -0.70710654 -0.70710654]
 [-0.21361012  0.21361012 -0.21360977  0.21360977  0.7071067  -0.7071067
   0.70710702 -0.70710702]
 [ 0.9637702   0.9637702   0.96377032  0.96377032  0.          0.          0.
   0.        ]
 [ 0.43523203 -0.43523203  0.43523188 -0.43523188 -0.09121676  0.09121676
  -0.09121681  0.09121681]
 [ 0.9430935   0.9430935  -0.94309347 -0.94309347  0.09121679  0.09121679
  -0.09121674 -0.09121674]
 [ 0.13688862 -0.13688862 -0.1368884   0.1368884  -0.64879876  0.64879876
   0.64879873 -0.64879873]]
TAM provided, skipping...
ThrusterManager: ready
Invalid output directory for the TAM matrix, dir= None
ThrusterManager: ready
[ INFO] [1518192969.574720255, 0.088000000]: Physics dynamic reconfigure ready.
[Wrn] [msgs.cc:1808] Conversion of sensor type[gps] not suppported.
[Wrn] [msgs.cc:1808] Conversion of sensor type[gps] not suppported.
[Wrn] [msgs.cc:1808] Conversion of sensor type[gps] not suppported.

musamarcusso, 2018-02-09:
Hi @phamhoanganhbk,

I think I found the problem (at least one of them). The controller node throws an exception at startup, an error can be read in this part:

```
from scipy.interpolate import splrep, splev
ImportError: No module named scipy.interpolate
```

Try installing `scipy` and `numpy` either with `apt` as follows

```
sudo apt-get install python-scipy python-numpy
```

or if you have `pip` (if you don't, you can install it using `sudo apt-get install python-pip`):

```
pip install scipy numpy
```

See if this solves the issue.

phamhoanganhbk, 2018-02-09:
Hi @musamarcusso 
Oh, here is the result :D. 
Merci beaucoup, 
 
[image]

musamarcusso, 2018-02-09:
Great that it worked out! 
I'm trying to check the issue with the visualization of the meshes also (the texture is not being loaded apparently on newer versions of Gazebo).
I'll close this issue now. If you have any other questions or bug reports, feel free to open a new issue here.

closed 2018-02-09 by musamarcusso