uuvsimulator/uuv_simulator #199: Can't launch custom actuated thruster actuated vehicle model

AditAg, 2018-02-10:
Hello,

I have been trying to launch my own bot and use it with the UUV_Simulator packages as specified in the Wiki. https://uuvsimulator.github.io/tutorials/config_thruster_manager.html. However when I launch the start_thruster_manager.launch file, I get the following errors:

gzserver: /home/adit/catkin_ws/src/uuv_simulator/uuv_gazebo_plugins/uuv_gazebo_plugins/src/UnderwaterObjectPlugin.cc:193: virtual void gazebo::UnderwaterObjectPlugin::Update(const gazebo::common::UpdateInfo&): Assertion `(!std::isnan(linearAccel) && !std::isnan(angularAccel))&&("Linear or angular accelerations are invalid.")' failed.
Aborted (core dumped)
[gazebo-2] process has died [pid 16890, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver --verbose -e ode worlds/empty_underwater.world __name:=gazebo __log:=/home/adit/.ros/log/f421e8be-0ea4-11e8-9a5c-ac2b6e309edb/gazebo-2.log].
log file: /home/adit/.ros/log/f421e8be-0ea4-11e8-9a5c-ac2b6e309edb/gazebo-2*.log

My files are attached here.

phamhoanganhbk, 2018-05-03:
Hi,
Coud you copy a step by step what you did?

noahl1, 2019-02-17:
Had a similar problem

JustBenj, 2019-02-18:
I just resolved this issue for myself. The problem was the propeller collision mesh intersecting with the base_link collision mesh. Make sure the position of your thruster is correct and that the scaling of your collision meshes are appropriate and you should have no issue. Hope this helps!