uuvsimulator/uuv_simulator #200: [Help me, Please]Problem with create a new thruster actuated vehicle model

phamhoanganhbk, 2018-02-12:
Hi everyone,

I have been following the tutorial to "creative a new thruster actuated vehicle model" and run 
`roslaunch uuv_descriptions_example upload_rov_example.launch`
then I have a problem as following  
______________________________________
> Traceback (most recent call last):
  File "/opt/ros/indigo/share/xacro/xacro.py", line 62, in <module>
    xacro.main()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 687, in main
    process_includes(doc, os.path.dirname(args[0]))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 269, in process_includes
    % (filename, str(e)))
xacro.XacroException: included file "/home/pha/catkin_ws/src/uuv_descriptions_example/models/rov_example/urdf/rov_example_base.xacro" generated an error during XML parsing: not well-formed (invalid token): line 135, column 12
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/indigo/share/xacro/xacro.py '/home/pha/catkin_ws/src/uuv_descriptions_example/models/rov_example/robots/rov_example_default.xacro' debug:=0 namespace:=rov_example] returned with code [1]. 

`Param xml is <param command="$(find xacro)/xacro.py '$(find uuv_descriptions_example)/models/rov_example/robots/rov_example_$(arg mode).xacro' debug:=$(arg debug) namespace:=$(arg namespace)" name="robot_description"/>
The traceback for the exception was written to the log file
` 
------------------------------------------------------------------------
I have tried by three ways :
+ I did everything like in the tutorial (This mean pkg name = uuv_descriptions_example)
+ Change the pkg name = abc_descriptions
+ Copy file in folder uuv_tutorial_rov_model
But I have the same problems as above.

Could you give me any ideas, any keywords to solve this problem, please?
Thank you for your help,

Here is the log file
[logfile.txt](https://github.com/uuvsimulator/uuv_simulator/files/1717948/logfile.txt)

swordsgnat, 2018-02-14:
Hey, man! I'm just a novice myself, but I have been working with the package for a few months. From my understanding, it's not the `/opt/ros/indigo/share/xacro/xacro.py` that's the issue - that's just a system-level file that's in charge of reading and interpreting xacro files. The issue is that `/home/pha/catkin_ws/src/uuv_descriptions_example/models/rov_example/urdf/rov_example_base.xacro` threw an error by having some weird symbol where it shouldn't have. That sounds like a typo of some sort to me, though I don't know where it is from this. Did you rewrite all of the rexrov files? I don't think you need to do that; I'm pretty sure I didn't when I did the tutorials. I'd try it just referencing the normal rexrov files and see if it works - if it does, then you know its something in your rewritten file that got wacky.

phamhoanganhbk, 2018-02-14:
Hi @swordsgnat ,
Many thanks for your suggest. In fact, I've just been working with ROS and AUV a few months too.
With the model rexrov, rexrov2 and desistek_saga (I cloned from github), they are running normally.
So, I want to make a new model to understanding eveything :D
I will check the file rov_example_base.xacro  again.
Best,

closed 2018-04-10 by phamhoanganhbk