uuvsimulator/uuv_simulator #202: The trajectory following tutorial

phamhoanganhbk, 2018-02-16:
Hi,
I am a newbie in auv and i am learning the trajectory following
https://github.com/uuvsimulator/uuv_simulator/wiki/Creating-a-custom-dynamic-positioning-controller

Could you please give me any documents/papers to explain the trajectory following algorithms which is used in the simulation program? Or a small tutorial of file `dp_controller_local_planner.py`, please.

Thank you for your help,

musamarcusso, 2018-02-21:
Hi @phamhoanganhbk,

the book used are reference to interpolate the waypoints and generate smooth paths can be found [here](https://www.springer.com/us/book/9783540856283).
Basically it consists in two interpolators, one that generates a [3D Bezier curves of third order](https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_control/uuv_trajectory_control/src/uuv_trajectory_generator/path_generator/cs_interpolator.py) and a [linear interpolator with polynomial blends](https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_control/uuv_trajectory_control/src/uuv_trajectory_generator/path_generator/lipb_interpolator.py).
If you want more information on Bezier curves, you can also check this website https://pomax.github.io/bezierinfo/.
I can also work on a documentation page soon, but at the moment I have had a lot of tasks on my list. Is there something more specific that you want to implement?

phamhoanganhbk, 2018-02-24:
Hi @musamarcusso 
Many thanks for your answer,
In fact, i want to implement formation tracking, application for uuv. then I would like to inherit your plugin to make the simulation.

closed 2018-02-28 by phamhoanganhbk