uuvsimulator/uuv_simulator #207: Problem running files described in "Creating a new custom dynamic positioning controller"

JuanNadales, 2018-02-21:
Hi Everyone,

I'm new to ROS and I'm trying to follow this tutorial. Everything works properly, but when running the command **roslaunch uuv_control_utils send_waypoints_file.launch uuv_name:=rexrov** I get the next error message :

[image]

Any idea why? Is there anything I should've done before that I haven't? I don't know either where to include the points which define the path the robot must follow. 

Thanks, hope you have time to answer soon.

swordsgnat, 2018-02-22:
Oh, man, check the glare on that screen! :D You can just copy/paste the error : P

Just to check: this is the "Creating a new custom dynamic positioning controller" tutorial, right, and you're running the line `roslaunch uuv_control_utils send_waypoints_file.launch uuv_name:=rexrov` at the same time another window is running `roslaunch uuv_tutorial_dp_controller start_tutorial_dp_controller_demo.launch`, right?

JuanNadales, 2018-02-22:
Thanks for the reply, and sorry for the glare, should've pasted the error message as you say :D 

When I executed the code I first ran  `roslaunch uuv_tutorial_dp_controller start_tutorial_dp_controller_demo.launch` and then in a new terminal I ran `roslaunch uuv_control_utils send_waypoints_file.launch uuv_name:=rexrov`. Should it be done the other way round?

swordsgnat, 2018-02-22:
No, that's right, I was just checking. I'm trying to reproduce the error on my machine, but I've been having difficulties with the tutorials myself since the Gazebo update to 7.10. I'll check on my other machine and get back to you. I'm just a beginner myself, so I don't know how much I can help, but it might give us a clue...

musamarcusso, 2018-02-23:
Hi @JuanNadales,

it can be that the controller node died before you started the `send_waypoints_file.launch`. Usually that is not that clear because Gazebo still runs, only the controller might have died. Could you copy here the output of you terminal when you run 

```
roslaunch uuv_tutorial_dp_controller start_tutorial_dp_controller_demo.launch
```

pxalcantara, 2020-05-19:
Hi @JuanNadales  is this issue still makes sense?