uuvsimulator/uuv_simulator #220: Sonar "-Inf" Issue

swordsgnat, 2018-03-21:
I recently updated to Gazebo 9.0 and the latest uuv_simulator version. I'm now having trouble with the sonar sensors (like the one included on rexrov_sonar). They're giving me "-Inf" values for all the ranges[], indicating that the ranges are less than the minimum or greater than the maximum, despite my positioning objects in the simulation to be in-range and adjusting the angle and placement of the sensor itself. The sensors were working fine before, though I'm not positive an update caused the change. Would anyone know why this might be happening, or should I be looking in the code for the sensors themselves?

brunohjs, 2018-05-06:
I have the same problem and I do not know what to do. Help us!

musamarcusso, 2018-05-08:
I will check this, I have Gazebo 9.0 running on my PC now. Could you just send me the configuration of you sonar plugin so that I can test the same problem?

YanielCarreno, 2018-12-26:
Hi there, 
I am now working with the uuv_simulator, and  I am facing the same error with the sonar sensor. I am using gazebo7 which is actually different to the version that you are working with. I am just wondering if  you find a way to fix the error.
Regards.

AndYanghlh, 2019-01-17:
Hi,@musamarcusso I am now facing this problem too ,and it really troubles me a lot. Have you found out the reason? thanks a lot

arturmiller, 2019-08-13:
Hello, I had the same issue with Gazebo7 and Gazebo9. In my case the problem was related to problems with the GPU.
I have found an issue where others had the same problem:
https://bitbucket.org/osrf/subt/issues/11/x2-laser-scanner-data-not-valid
A fix, which worked for me is to use the CPU instead of the GPU (performance will decrease). For that you have to change to lines: https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_sensor_plugins/uuv_sensor_ros_plugins/urdf/sonar_snippets.xacro
line 130: "gpu_ray" -> "ray"
line 154: "libgazebo_ros_gpu_laser.so" -> "libgazebo_ros_laser.so"

ccs-ros, 2019-10-10:
Hi, @arturmiller I used your methord with Gazebo7,and the problem still exists.And here are some warnings:
[Wrn] [msgs.cc:1655] Conversion of sensor type[gps] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[gpu_ray] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[gps] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[gpu_ray] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[gps] not suppported.
[Wrn] [msgs.cc:1655] Conversion of sensor type[gpu_ray] not suppported.
could you help me ? thanks a lot!