uuvsimulator/uuv_simulator #230: How to get Euler angle ??

phamhoanganhbk, 2018-04-24:
Hi everyone,

I am trying to get the Euler angle to display the Matlab/Simulink.
I used the topic `/rexrov/pose_gt` to have an orientation [x y z w] then i convert it to Euler angle  but seemly it isnot correct. I controlled uuv revolving, but the yaw angle is always 3.14 (Like the image I attached below).

Could you give me some suggests, please?
[image]

Thank you very much.

musamarcusso, 2018-05-03:
Hi @phamhoanganhbk, sorry for the late response, could you copy some samples of the `pose_gt` output and the data you are reading in Matlab?

phamhoanganhbk, 2018-05-03:
Hi @musamarcusso ,
Thank you for your reply.
I found an error for this issus. I am sorry for not closing this topic.
Have a nice day.

closed 2018-05-03 by phamhoanganhbk