uuvsimulator/uuv_simulator #243: A question about the sonar data

BeyondLeee, 2018-05-08:
Hi, I want to study the obstacle avoidance problem of multi robots and I need  the multibeam sonar data of the rexrov.  I saw in the paper that the UAV screenshot with sonar is like this
[image]
but in my project ,I didn't the blue lasers! Could you tell me how to find them out? And What is the topic of publishing sonar data? Is  /rexrov/dvl_sonar0?
Thank you very much!

musamarcusso, 2018-07-02:
Hi @BeyondLeee, the sonar output for Gazebo has been disabled for performance reasons, but it is configurable through [this option](https://github.com/uuvsimulator/uuv_simulator/blob/ff3fc7fa5123fc9699846534e6d6e81ef78d426d/uuv_sensor_plugins/uuv_sensor_ros_plugins/urdf/sonar_snippets.xacro#L132) of the Gazebo plugin. The sonar simulation is still a feature to be developed, so for now the laser scanner plugins are being used. 
You might, however, have a better visualization in RViz, have you tried that?

pxalcantara, 2020-05-19:
Hi @BeyondLeee  is this issue still makes sense?