uuvsimulator/uuv_simulator #246: Including vehicle into the world

hbaqueiro, 2018-05-11:
Hi everyone,

(I'm a beginner to ROS, Gazebo and UUV Simulator) I am doing the tutorial of UUV Simulator and in the section "Configuring the thruster manager for a new vehicle" I am getting the following error when I try to include my vehicle into the world:
`baqueiro@baqueiro-Latitude-E6430:~/catkin_ws$ roslaunch baqtronic_description upload.launch 
... logging to /home/baqueiro/.ros/log/136ecaac-5522-11e8-9ca6-f01faf1d30a4/roslaunch-baqueiro-Latitude-E6430-12866.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
XML parsing error: not well-formed (invalid token): line 4, column 8
when processing file: /home/baqueiro/catkin_ws/src/baqtronic_description/urdf/snippets.xacro
included from: /home/baqueiro/catkin_ws/src/baqtronic_description/urdf/base.xacro
included from: /home/baqueiro/catkin_ws/src/baqtronic_description/robots/default.xacro

Check that:
 - Your XML is well-formed
 - You have the xacro xmlns declaration: xmlns:xacro="http://www.ros.org/wiki/xacro"
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/share/xacro/xacro.py '/home/baqueiro/catkin_ws/src/baqtronic_description/robots/default.xacro' debug:=0 namespace:=baqtronic] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro.py '$(find baqtronic_description)/robots/$(arg mode).xacro' debug:=$(arg debug) namespace:=$(arg namespace)" name="robot_description"/>
The traceback for the exception was written to the log file`

Can anyone help me? Thanks in advance :)

musamarcusso, 2018-07-02:
Hi @hbaqueiro, could you provide the URDF of your robot model as well?

hbaqueiro, 2018-07-13:
Hi @musamarcusso , sorry for the delay. I've formatted my machine and hat to follow the tutorial again. The URDF of my robot are the templates generated by ```rosrun uuv_assistants create_new_robot_model --robot_name baqfish```. (I named it baqfish instead of baqtronic this time). I've not edited them. I could paste the code but they should be very much the same as if you run the command...

brschettini, 2018-07-18:
Hello everyone, 

I'm also learning the uuv_simulator and went to the same problem. It appear to be a problem on the template's comment. Somehow it's not allowing comments like the one below:

```          <!--
            ----------------------------------------------------
            ----------------------------------------------------
            Further parameters necessary for the lift and drag plugin
            ----------------------------------------------------
            ----------------------------------------------------
          -->```

Also, I realized that a suffix tag in the sensors.template was missing.

I made a pull request to fix those problems.

hbaqueiro, 2018-07-23:
Dear all,
I updated the autoproj to get the fixed code and created the new robot model and  the thruster manager configuration again. I am still facing errors:
```xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/xacro/xacro.py", line 61, in <module>
    xacro.main()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/xacro/__init__.py", line 1073, in main
    out.write(doc.toprettyxml(indent='  '))
UnicodeEncodeError: 'ascii' codec can't encode character u'\u201c' in position 10790: ordinal not in range(128)
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/share/xacro/xacro.py '/home/baqueiro/autoproj/src/baqfish_description/robots/default.xacro' debug:=0 namespace:=baqfish] returned with code [1]. 

Param xml is <param command="$(find xacro)/xacro.py '$(find baqfish_description)/robots/$(arg mode).xacro' debug:=$(arg debug) namespace:=$(arg namespace)" name="robot_description"/>
The traceback for the exception was written to the log file
```

pxalcantara, 2020-05-19:
Hi @hbaqueiro  is this issue still makes sense? Can we close it?

closed 2020-05-20 by hbaqueiro