uuvsimulator/uuv_simulator #247: rexrov demo teleop throws error(s)

james-gonz0, 2018-06-15:
So I am trying to run the quickstart, and I get past running gazebo with empty_underwater_world and inserting the ROV into said world, however when I run the teleop command line, i get the following errors:

`[rexrov/thruster_allocator-1] process has died [pid 27523, exit code 1, cmd /home/cavr/catkin_ws/src/uuvsimulator/uuv_control/uuv_thruster_manager/scripts/thruster_allocator.py __name:=thruster_allocator __log:=/home/cavr/.ros/log/a398ff82-70b9-11e8-b03e-54ee75532f16/rexrov-thruster_allocator-1.log].
log file: /home/cavr/.ros/log/a398ff82-70b9-11e8-b03e-54ee75532f16/rexrov-thruster_allocator-1*.log

[rexrov/velocity_control-3] process has died [pid 27525, exit code 1, cmd /home/cavr/catkin_ws/src/uuvsimulator/uuv_control/uuv_control_cascaded_pids/scripts/VelocityControl.py odom:=/rexrov/pose_gt cmd_accel:=/rexrov/cmd_accel __name:=velocity_control __log:=/home/cavr/.ros/log/a398ff82-70b9-11e8-b03e-54ee75532f16/rexrov-velocity_control-3.log].
log file: /home/cavr/.ros/log/a398ff82-70b9-11e8-b03e-54ee75532f16/rexrov-velocity_control-3*.log
`
Thanks in advance, any help would be greatly appreciated.

musamarcusso, 2018-06-18:
Hi @james-gonz0,
could you send me the complete output? I can't really see what caused the problem.

james-gonz0, 2018-06-18:
Here it is:
`roslaunch uuv_control_cascaded_pid joy_velocity.launch uuv_name:=rexrov model_name:=rexrov joy_id:=0
cavr@hab2:~$ roslaunch uuv_control_cascaded_pid joy_velocity.launch uuv_name:=rexrov model_name:=rexrov joy_id:=0
... logging to /home/cavr/.ros/log/02570ab0-7311-11e8-b639-54ee75532f16/roslaunch-hab2-4754.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hab2:42812/

SUMMARY
========

PARAMETERS
 * /rexrov/acceleration_control/tf_prefix: rexrov/
 * /rexrov/joy_uuv_velocity_teleop/deadman_button: -1
 * /rexrov/joy_uuv_velocity_teleop/exclusion_buttons: [4, 5]
 * /rexrov/joy_uuv_velocity_teleop/mapping/pitch/axis: -1
 * /rexrov/joy_uuv_velocity_teleop/mapping/pitch/gain: 0.0
 * /rexrov/joy_uuv_velocity_teleop/mapping/roll/axis: -1
 * /rexrov/joy_uuv_velocity_teleop/mapping/roll/gain: 0.0
 * /rexrov/joy_uuv_velocity_teleop/mapping/x/axis: 4
 * /rexrov/joy_uuv_velocity_teleop/mapping/x/gain: 0.3
 * /rexrov/joy_uuv_velocity_teleop/mapping/y/axis: 3
 * /rexrov/joy_uuv_velocity_teleop/mapping/y/gain: 0.3
 * /rexrov/joy_uuv_velocity_teleop/mapping/yaw/axis: 0
 * /rexrov/joy_uuv_velocity_teleop/mapping/yaw/gain: 0.2
 * /rexrov/joy_uuv_velocity_teleop/mapping/z/axis: 1
 * /rexrov/joy_uuv_velocity_teleop/mapping/z/gain: 0.3
 * /rexrov/joy_uuv_velocity_teleop/type: twist
 * /rexrov/joystick/autorepeat_rate: 10
 * /rexrov/joystick/dev: /dev/input/js0
 * /rexrov/pid/inertial/ixx: 525.39
 * /rexrov/pid/inertial/ixy: 0
 * /rexrov/pid/inertial/ixz: 0
 * /rexrov/pid/inertial/iyy: 794.2
 * /rexrov/pid/inertial/iyz: 0
 * /rexrov/pid/inertial/izz: 691.23
 * /rexrov/pid/mass: 1862.0
 * /rexrov/thruster_allocator/tam: [[0.1597426884844...
 * /rexrov/thruster_manager/base_link: base_link
 * /rexrov/thruster_manager/conversion_fcn: proportional
 * /rexrov/thruster_manager/conversion_fcn_params/gain: 0.00031
 * /rexrov/thruster_manager/max_thrust: 2000.0
 * /rexrov/thruster_manager/tf_prefix: rexrov
 * /rexrov/thruster_manager/thruster_frame_base: thruster_
 * /rexrov/thruster_manager/thruster_topic_prefix: thrusters/
 * /rexrov/thruster_manager/thruster_topic_suffix: /input
 * /rexrov/thruster_manager/timeout: -1
 * /rexrov/thruster_manager/update_rate: 50
 * /rexrov/velocity_control/angular_d: 1.0
 * /rexrov/velocity_control/angular_i: 2.0
 * /rexrov/velocity_control/angular_p: 10.0
 * /rexrov/velocity_control/angular_sat: 5.0
 * /rexrov/velocity_control/linear_d: 0.0
 * /rexrov/velocity_control/linear_i: 2.0
 * /rexrov/velocity_control/linear_p: 10.0
 * /rexrov/velocity_control/linear_sat: 20.0
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /rexrov/
    acceleration_control (uuv_control_cascaded_pid/AccelerationControl.py)
    joy_uuv_velocity_teleop (uuv_teleop/vehicle_teleop.py)
    joystick (joy/joy_node)
    thruster_allocator (uuv_thruster_manager/thruster_allocator.py)
    velocity_control (uuv_control_cascaded_pid/VelocityControl.py)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[rexrov/thruster_allocator-1]: started with pid [4774]
process[rexrov/acceleration_control-2]: started with pid [4775]
process[rexrov/velocity_control-3]: started with pid [4776]
process[rexrov/joystick-4]: started with pid [4777]
process[rexrov/joy_uuv_velocity_teleop-5]: started with pid [4778]
starting AccelerationControl.py
Traceback (most recent call last):
  File "/home/cavr/catkin_ws/src/uuvsimulator/uuv_control/uuv_thruster_manager/scripts/thruster_allocator.py", line 25, in <module>
    from uuv_thrusters import ThrusterManager
ImportError: No module named uuv_thrusters
AccelerationControllerNode: initializing node
[[ 1862.       0.       0.       0.       0.       0.  ]
 [    0.    1862.       0.       0.       0.       0.  ]
 [    0.       0.    1862.       0.       0.       0.  ]
 [    0.       0.       0.     525.39     0.       0.  ]
 [    0.       0.       0.       0.     794.2      0.  ]
 [    0.       0.       0.       0.       0.     691.23]]
Traceback (most recent call last):
  File "/home/cavr/catkin_ws/src/uuvsimulator/uuv_control/uuv_control_cascaded_pids/scripts/VelocityControl.py", line 26, in <module>
    from PID import PIDRegulator
ImportError: No module named PID
[rexrov/thruster_allocator-1] process has died [pid 4774, exit code 1, cmd /home/cavr/catkin_ws/src/uuvsimulator/uuv_control/uuv_thruster_manager/scripts/thruster_allocator.py __name:=thruster_allocator __log:=/home/cavr/.ros/log/02570ab0-7311-11e8-b639-54ee75532f16/rexrov-thruster_allocator-1.log].
log file: /home/cavr/.ros/log/02570ab0-7311-11e8-b639-54ee75532f16/rexrov-thruster_allocator-1*.log
[rexrov/velocity_control-3] process has died [pid 4776, exit code 1, cmd /home/cavr/catkin_ws/src/uuvsimulator/uuv_control/uuv_control_cascaded_pids/scripts/VelocityControl.py odom:=/rexrov/pose_gt cmd_accel:=/rexrov/cmd_accel __name:=velocity_control __log:=/home/cavr/.ros/log/02570ab0-7311-11e8-b639-54ee75532f16/rexrov-velocity_control-3.log].
log file: /home/cavr/.ros/log/02570ab0-7311-11e8-b639-54ee75532f16/rexrov-velocity_control-3*.log

`

james-gonz0, 2018-06-29:
Resolved, was actually a catkin_ws error relating to cmake

musamarcusso, 2018-07-02:
Hi @james-gonz0, great that it is resolved. May I close this issue then?

james-gonz0, 2018-07-02:
@musamarcusso Yes you may.

closed 2018-07-03 by musamarcusso