uuvsimulator/uuv_simulator #254: Newly added spherical robot keeps jumping in and out of water

touhid1259, 2018-07-03:
Hi

I am new to gazebo, ROS and UUV simulator.  I am trying to use UUV Simulator. I am trying to add new robot according to your documentation. I have added one sphere robot with radius .2. But, the robot is jumping up and down of water (Images attached). Actually, my expectation was:- robot may sink or float or may stays stable inside water based on its mass. So, can you please help me on this regard. Why the robot is keeps jumping up and down inside water? 

sphere robot parameters are:
mass: 30 (I have tweaked it around with high and low values. Sometime it sinks which is okay but the above issue is my problem)
radius: .2
volume: 0.0335 (using the equation:  4/3 * pi * radius * radius * radius)
center of gravity co-ordinate: 0 0 0
center of buoyancy co-ordinate: 0 0 0.01 
and fluid density is: 1028

[image]
[image]
[image]

musamarcusso, 2018-07-03:
Hi @touhid1259, from a first glance in this problem, it seems to me that the damping coefficients may not be enough to slow down the ball. 
Could you copy the complete model you are using here? Then it might be easier to evaluate what to do :)

touhid1259, 2018-07-03:
Hi musamarcusso

Thanks for the reply. Below are the generated robot files that are being generated following the documentation. I have changed the files to my need. In my case, I think only default.xacro, gazebo.xacro, base.xacro and sensors.xacro are used. But I added all the files. One thing to note, from gazebo.xacro, I have used the hydrodynamic model for sphere instead of fossen. So, may be that is the issue here, as I can see that the damping coefficients generating here. Another thing is, when I ran the upload.launch for the robot I get the below message regarding added_mass and damping coefficients:

```XML
ea_robot::ea_robot/base_link::added_mass
     17.2243           0           0           0           0           0
           0     17.2243           0           0           0           0
           0           0     17.2243           0           0           0
           0           0           0           0           0           0
           0           0           0           0           0           0
           0           0           0           0           0           0
ea_robot::ea_robot/base_link::scaling_added_mass
ea_robot::ea_robot/base_link::scaling_added_mass
ea_robot::ea_robot/base_link::offset_added_mass
ea_robot::ea_robot/base_link::offset_added_mass
ea_robot::ea_robot/base_link::linear_damping
ea_robot::ea_robot/base_link::linear_damping
           0           0           0           0           0           0
           0           0           0           0           0           0
           0           0           0           0           0           0
           0           0           0           0           0           0
           0           0           0           0           0           0
           0           0           0           0           0           0
ea_robot::ea_robot/base_link::linear_damping_forward_speed
ea_robot::ea_robot/base_link::linear_damping_forward_speed
           0           0           0           0           0           0
           0           0           0           0           0           0
           0           0           0           0           0           0
           0           0           0           0           0           0
           0           0           0           0           0           0
           0           0           0           0           0           0
ea_robot::ea_robot/base_link::quadratic_damping
ea_robot::ea_robot/base_link::quadratic_damping
           0           0           0           0           0           0
           0           0           0           0           0           0
           0           0           0           0           0           0
           0           0           0           0           0           0
           0           0           0           0           0           0
           0           0           0           0           0           0
 ```

**default.xacro**
```XML
<?xml version="1.0"?>
<robot name="ea_robot" xmlns:xacro="http://www.ros.org/wiki/xacro" >
  <xacro:arg name="debug" default="0"/>
  <xacro:arg name="namespace" default="ea_robot"/>
  
  <!-- ea_robot's genome related arguments -->	        
  <xacro:arg name="box1GenomeBit" default= '0' />
  <xacro:arg name="box2GenomeBit" default= '0' />
  <xacro:arg name="box3GenomeBit" default= '0' />
  <xacro:arg name="box4GenomeBit" default= '0' />
  <xacro:arg name="box5GenomeBit" default= '0' />
  <xacro:arg name="box6GenomeBit" default= '0' />
  <xacro:arg name="box7GenomeBit" default= '0' />
  <xacro:arg name="box8GenomeBit" default= '0' />
  
  <!-- Include the ROV macro file -->
  <xacro:include filename="$(find ea_robot_description)/urdf/base.xacro"/>
  <xacro:include filename="$(find ea_robot_description)/urdf/gazebo.xacro"/>

  <!-- Create the ea_robot -->
  <xacro:ea_robot_base namespace="$(arg namespace)" box1GenomeBit="$(arg box1GenomeBit)" box2GenomeBit="$(arg box2GenomeBit)"
  					box3GenomeBit="$(arg box3GenomeBit)" box4GenomeBit="$(arg box4GenomeBit)"
  					box5GenomeBit="$(arg box5GenomeBit)" box6GenomeBit="$(arg box6GenomeBit)"
  					box7GenomeBit="$(arg box7GenomeBit)" box8GenomeBit="$(arg box8GenomeBit)" >
    <!-- The underwater object plugin is given as an input block parameter to
     allow the addition of external models of manipulator units -->
[... 728 lines cut ...]

          <!--

            Further parameters necessary for the lift and drag plugin

          -->

          <!-- Current velocity topic -->
          <current_velocity_topic>/hydrodynamics/current_velocity</current_velocity_topic>

          <!-- Name of the correspodent fin link and joint  -->
          <link_name>${namespace}/fin${fin_id}</link_name>
          <joint_name>${namespace}/fin${fin_id}_joint</joint_name>

          <!-- Output topic to publish the current angle of the fin joint -->
          <output_topic>${namespace}/fins/${fin_id}/output</output_topic>
          <input_topic>${namespace}/fins/${fin_id}/input</input_topic>
          <wrench_topic>${namespace}/fins/${fin_id}/wrench_topic</wrench_topic>
        </plugin>
      </gazebo>
  </xacro:macro>

</robot>

```

musamarcusso, 2018-07-05:
Hi @touhid1259, thanks for the information! I found a bug on the computation of the nonlinear damping coefficients for the sphere model related to a refactoring that happened a while ago. I am working on a patch and will notify you when it is finished.

musamarcusso, 2018-07-05:
Hi @touhid1259, the damping problem should be solved for the sphere computation, could you check if it works?

touhid1259, 2018-07-06:
Hi musamarcusso

After I pulling the new code, I am getting an error for `catkin_make install` command. The error is:-

```
CMake Error at uuv_simulator/uuv_gazebo_plugins/uuv_gazebo_plugins/cmake_install.cmake:179 (file):
  file INSTALL cannot find
  "/home/touhid/catkin_ws_UUV/src/uuv_simulator/uuv_gazebo_plugins/uuv_gazebo_plugins/test".
Call Stack (most recent call first):
  cmake_install.cmake:135 (include)
  

Makefile:61: recipe for target 'install' failed
make: *** [install] Error 1
Invoking "make install -j4 -l4" failed
```

So, do I have to clone the repository again? Please, provide me any suggestion.

Thanks
Touhid

musamarcusso, 2018-07-07:
Hi @touhid1259, I found the mistake and will put a patch to that soon. I usually use the workspace configuration described in the [documentation](https://uuvsimulator.github.io/installation.html#creating-and-configuring-a-workspace) and therefore the error didn't show up to me.

musamarcusso, 2018-07-09:
Hi @touhid1259, I recently merged the fix for that issue, do you still see the same error?

touhid1259, 2018-07-09:
Hi musamarcusso

Thanks for the fix. It is working perfectly now. Also, the robot is working as expected. I have couple of questions though. These are:

1. Can you please tell me how am I suppose to calculate the "hydrodynamic parameters"  i.e., added_mass and damping parameters for different kind of robots i.e., robots other than "spherical shape" or the added robots in the tutorial?

2. Can I accelerate the simulation in gazebo i.e., simulating "10 mins in real time" with 2 mins?

I know #2 doesn't really concern this uuv simulator tool. Actually, I am trying to achieve that with uuv simulator and googled a bit to know more about it. But still not convinced or rather say fully understand about accelerating real time in simulation in gazebo. So, I am just asking for suggestions. It will be really helpful for me if you could provide some suggestions about these two questions.

And again thanks for the hotfix. It is working smoothly now.

Thanks
Touhid

musamarcusso, 2018-07-10:
Hi @touhid1259, I am glad that the issue is solved.
Regarding your questions:
1. There are ways of calculating the added-mass and damping coefficients by using approximations to simple shapes (for example, cylinders, spheres and so on). A few sources come to mind for me where you might find the correct equations to do that:
* [Values of the Added Mass](http://www.brennen.caltech.edu/fluidbook/basicfluiddynamics/unsteadyflows/addedmass/valuesoftheaddedmass.pdf)
* [DNV Environmental Conditions and Environmental Loads](https://rules.dnvgl.com/docs/pdf/DNV/codes/docs/2010-10/RP-C205.pdf) - have a look into Appendix D and E
* [MIT OpenCourseWare Hydrodynamics (13.012)](https://ocw.mit.edu/courses/mechanical-engineering/2-016-hydrodynamics-13-012-fall-2005/readings/)
* [MIT OpenCourseWare Marine Hydrodynamics (13.021)](https://ocw.mit.edu/courses/mechanical-engineering/2-20-marine-hydrodynamics-13-021-spring-2005/lecture-notes/)

The spherical shape is simple but there is no damping in roll, pitch and yaw. Therefore it might lead to some unwanted behavior. But I think you will find more derivations in the material above.

2. You can accelerate the simulation by changing [this parameter](https://github.com/uuvsimulator/uuv_simulator/blob/c04e9f4787886297ed08796863f77eb716e660d2/uuv_descriptions/worlds/mangalia.world#L21) in the world file, or go to the Gazebo GUI on the left side, click on "Physics" and change the parameters for "real time update" and "max step size". Be careful that the simulation can start to become unstable when you do that, depending on the parameters you are using.

touhid1259, 2018-07-10:
Hi musamarcusso

Thanks for the suggestions.

regards
touhid

closed 2018-07-10 by musamarcusso