uuvsimulator/uuv_simulator #255: Issues with following the dp controll tutorial

dbiegl, 2018-07-03:
Im trying to run the dynamic position simulation but whenever I run -- roslaunch uuv_tutorial_dp_controller start_tutorial_dp_controller_demo.launch -- I get the error -- [rexrov/tutorial_dp_controller-11] process has died [pid 8881, exit code 1, cmd /home/daniel/catkin_ws/src/uuv_simulator/uuv_tutorials/uuv_tutorial_dp_controller/scripts/tutorial_dp_controller.py odom:=pose_gt trajectory:=dp_controller/trajectory input_trajectory:=dp_controller/input_trajectory waypoints:=dp_controller/waypoints error:=dp_controller/error reference:=dp_controller/reference thruster_output:=thruster_manager/input_stamped __name:=tutorial_dp_controller __log:=/home/daniel/.ros/log/7d6c0026-7eeb-11e8-91a4-b8ca3a8b9ea8/rexrov-tutorial_dp_controller-11.log].
log file: /home/daniel/.ros/log/7d6c0026-7eeb-11e8-91a4-b8ca3a8b9ea8/rexrov-tutorial_dp_controller-11*.log -- gazebo and rviz launch normally but when I try to run -- roslaunch uuv_control_utils send_waypoints_file.launch uuv_name:=rexrov -- in another terminal I get the error -- [ERROR] [1530641675.319500, 0.000000]: Negative start time, setting it to 0.0 -- thanks for any help you can offer

dbiegl, 2018-07-05:
After a nice day off yesterday, I installed casadi and everything worked perfectly. Im closing the issue.

closed 2018-07-05 by dbiegl

musamarcusso, 2018-07-05:
Hi @dbiegl, thanks for the feedback. I will add the installation of casadi to the list of requirements in the documentation.