uuvsimulator/uuv_simulator #260: Trying to launch eca_a9 causes errors.

dbiegl, 2018-07-05:
After getting rexrov to follow waypoints im trying to do the same with eca_a9. I launch a world just fine, but when i try to spawn my rover with ``roslaunch eca_a9_gazebo start_demo_teleop.launch joy_id:=0``
I get the this output.......
logging to /home/daniel/.ros/log/a49d83de-808c-11e8-9226-b8ca3a8b9ea8/roslaunch-daniel-Precision-T1650-1513.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Undefined parameters [noise_sigma,noise_amplitude] None None
when instantiating macro: dvl_plugin_macro (/home/daniel/catkin_ws/src/uuv_simulator/uuv_sensor_plugins/uuv_sensor_ros_plugins/urdf/dvl_snippets.xacro)
instantiated from: eca_a9_base (/home/daniel/catkin_ws/src/eca_a9/eca_a9_description/urdf/eca_a9_base.xacro)
in file: /home/daniel/catkin_ws/src/eca_a9/eca_a9_description/urdf/eca_a9_sensors.xacro
included from: /home/daniel/catkin_ws/src/eca_a9/eca_a9_description/robots/eca_a9_default.urdf.xacro
while processing /home/daniel/catkin_ws/src/eca_a9/eca_a9_description/launch/upload_eca_a9.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro --inorder '/home/daniel/catkin_ws/src/eca_a9/eca_a9_description/robots/eca_a9_default.urdf.xacro' debug:=0 namespace:=eca_a9] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro --inorder '$(find eca_a9_description)/robots/eca_a9_$(arg mode).urdf.xacro' debug:=$(arg debug) namespace:=$(arg namespace)" name="robot_description"/>
The traceback for the exception was written to the log file

any help is appreciated. Thanks

musamarcusso, 2018-07-05:
Hi @dbiegl, I have a look into this. I am actually currently working on the ECA A9 package to update it to the latest changes.

musamarcusso, 2018-07-05:
If you are looking for another AUV model, you can try the [LAUV](https://github.com/uuvsimulator/lauv_gazebo) as well.
I will get back to you when the issue with the ECA A9 are solved.

dbiegl, 2018-07-05:
thank you for your swift response, Ill try to use LAUV for now and close this issue.

closed 2018-07-05 by dbiegl

musamarcusso, 2018-07-05:
@dbiegl I just merged a patch to the ECA A9 repository. The error seems to be solved on my end, could you check if you still have problems?

musamarcusso, 2018-07-06:
Thanks for the feedback!