uuvsimulator/uuv_simulator #272: Creating a new custom dynamic positioning controller

galfunk, 2018-07-09:
Hello try to run 
roslaunch uuv_tutorial_dp_controller start_tutorial_dp_controller_demo.launch

the following error appears

[INFO] [1531160226.812975, 0.000000]: Calling service /gazebo/spawn_urdf_model
[rexrov/tutorial_dp_controller-11] process has died [pid 22836, exit code 1, cmd /home/gal/catkin_ws/src/uuv_simulator/uuv_tutorials/uuv_tutorial_dp_controller/scripts/tutorial_dp_controller.py odom:=pose_gt trajectory:=dp_controller/trajectory input_trajectory:=dp_controller/input_trajectory waypoints:=dp_controller/waypoints error:=dp_controller/error reference:=dp_controller/reference thruster_output:=thruster_manager/input_stamped __name:=tutorial_dp_controller __log:=/home/gal/.ros/log/486a11ac-83a4-11e8-ac53-9cb6d0f2f06b/rexrov-tutorial_dp_controller-11.log].
log file: /home/gal/.ros/log/486a11ac-83a4-11e8-ac53-9cb6d0f2f06b/rexrov-tutorial_dp_controller-11*.log

any help??

Many thanks 

Gal

musamarcusso, 2018-07-10:
Hi @galfunk, could you provide the whole output where this error occurs?

galfunk, 2018-07-10:
Thank you for your reply, please find the whole output 

gal@gal-XPS-15-9560:~$ roslaunch eca_a9_gazebo start_demo_teleop.launch joy_id:=0
... logging to /home/gal/.ros/log/ba51bd2c-83d8-11e8-ac53-9cb6d0f2f06b/roslaunch-gal-XPS-15-9560-19453.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global property: gamma
when processing file: /home/gal/catkin/src/eca_a9/eca_a9_description/urdf/eca_a9_sensors.xacro
included from: /home/gal/catkin/src/eca_a9/eca_a9_description/robots/eca_a9_default.urdf.xacro
started roslaunch server http://gal-XPS-15-9560:45909/

SUMMARY
========

PARAMETERS
 * /eca_a9/finned_uuv_teleop/axis_pitch: 4
 * /eca_a9/finned_uuv_teleop/axis_roll: 0
 * /eca_a9/finned_uuv_teleop/axis_thruster: 1
 * /eca_a9/finned_uuv_teleop/axis_yaw: 3
 * /eca_a9/finned_uuv_teleop/fin_topic_prefix: fins/
 * /eca_a9/finned_uuv_teleop/fin_topic_suffix: /input
 * /eca_a9/finned_uuv_teleop/gain_pitch: [1, 1, -1, -1]
 * /eca_a9/finned_uuv_teleop/gain_roll: [1, 1, 1, 1]
 * /eca_a9/finned_uuv_teleop/gain_yaw: [-1, 1, 1, -1]
 * /eca_a9/finned_uuv_teleop/n_fins: 4
 * /eca_a9/finned_uuv_teleop/thruster_joy_gain: -1.0
 * /eca_a9/finned_uuv_teleop/thruster_model/max_thrust: 5
 * /eca_a9/finned_uuv_teleop/thruster_model/name: proportional
 * /eca_a9/finned_uuv_teleop/thruster_model/params/gain: 4.9e-05
 * /eca_a9/finned_uuv_teleop/thruster_topic: thrusters/0/input
 * /eca_a9/joystick/autorepeat_rate: 10
 * /eca_a9/joystick/dev: /dev/input/js0
 * /eca_a9/robot_description: <?xml version="1....
 * /eca_a9/robot_state_publisher/publish_frequency: 5
 * /eca_a9/robot_state_publisher/robot_description: /eca_a9/robot_des...
 * /ground_truth_to_tf_eca_a9/child_frame_id: /eca_a9/base_link
 * /ground_truth_to_tf_eca_a9/footprint_frame_id: /eca_a9/base_foot...
 * /ground_truth_to_tf_eca_a9/frame_id: /world
 * /ground_truth_to_tf_eca_a9/odometry_topic: /eca_a9/pose_gt
 * /ground_truth_to_tf_eca_a9/stabilized_frame_id: /eca_a9/base_stab...
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /eca_a9/
    finned_uuv_teleop (uuv_teleop/finned_uuv_teleop.py)
    joystick (joy/joy_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    urdf_spawner (uuv_descriptions/spawn_model.py)
  /
    ground_truth_to_tf_eca_a9 (message_to_tf/message_to_tf)
    recording (rosbag/record)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[eca_a9/urdf_spawner-1]: started with pid [19474]
process[eca_a9/robot_state_publisher-2]: started with pid [19475]
process[ground_truth_to_tf_eca_a9-3]: started with pid [19476]
process[eca_a9/finned_uuv_teleop-4]: started with pid [19483]
process[eca_a9/joystick-5]: started with pid [19500]
[ WARN] [1531182762.513436604]: The root link eca_a9/base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
process[rviz-6]: started with pid [19524]
process[recording-7]: started with pid [19536]
[ INFO] [1531182762.725779339]: rviz version 1.12.16
[ INFO] [1531182762.725864979]: compiled against Qt version 5.5.1
[ INFO] [1531182762.725880571]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1531182762.937683709]: Stereo is NOT SUPPORTED
[... 352 lines cut ...]
Thruster #7 - proportional - thrusters/7/input
[INFO] [1531214086.175534, 0.082000]: Thruster model:
[INFO] [1531214086.175823, 0.082000]: 	Gain=0.00031
[INFO] [1531214086.176189, 0.084000]: Thruster allocation matrix provided!
[INFO] [1531214086.176417, 0.084000]: TAM=
[INFO] [1531214086.177788, 0.084000]: [[ 0.15974269  0.15974269 -0.15974243 -0.15974243  0.70710686  0.70710686
  -0.70710654 -0.70710654]
 [-0.21361012  0.21361012 -0.21360977  0.21360977  0.7071067  -0.7071067
   0.70710702 -0.70710702]
 [ 0.9637702   0.9637702   0.96377032  0.96377032  0.          0.          0.
   0.        ]
 [ 0.43523203 -0.43523203  0.43523188 -0.43523188 -0.09121676  0.09121676
  -0.09121681  0.09121681]
 [ 0.9430935   0.9430935  -0.94309347 -0.94309347  0.09121679  0.09121679
  -0.09121674 -0.09121674]
 [ 0.13688862 -0.13688862 -0.1368884   0.1368884  -0.64879876  0.64879876
   0.64879873 -0.64879873]]
[ INFO] [1531214086.179505844, 0.086000000]: Physics dynamic reconfigure ready.
[INFO] [1531214086.181852, 0.088000]: TAM provided, skipping...
[INFO] [1531214086.182160, 0.088000]: ThrusterManager: ready
[INFO] [1531214086.182757, 0.090000]: Invalid output directory for the TAM matrix, dir=None
[INFO] [1531214086.183083, 0.090000]: ThrusterManager: ready
[ INFO] [1531214086.183410703, 0.090000000]: Physics dynamic reconfigure ready.
[Wrn] [msgs.cc:1808] Conversion of sensor type[gps] not suppported.
[Wrn] [msgs.cc:1808] Conversion of sensor type[gps] not suppported.

musamarcusso, 2018-07-10:
Hi @galfunk, thanks for the information.
It seems that you are missing `casadi` in your system. This is part of a new implementation but it shouldn't affect other controllers. I will fix this to enforce the use of this package to be optional.
If you want to fix this now, you can install `casadi` using 

```
sudo pip install casadi
```

galfunk, 2018-07-10:
I have another question regarding the option to generate a ROS bag with the recorded simulated data, if I set the flag "true", where the location of the recorded bag file define ?

Thanks again

galfunk, 2018-07-10:
I try to install the casadi - unfortunately received sudo: pip: command not found

musamarcusso, 2018-07-10:
For `pip`: to install `pip` you have to run 

```
sudo apt install python-pip
```

In the tutorial I forgot to put the option to determine the path to the `recording.bag` file. Per default it should be stored in your `$HOME/.ros` folder. If you check the [`record.launch` file](https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_gazebo/launch/controller_demos/record_demo.launch#L3), you should be able to see the argument `bag_filename` with the full path for the ROS bag. I will add option this to the tutorial, though.