uuvsimulator/uuv_simulator #274: uuv_control_utils package

galfunk, 2018-07-10:
Hi , try to run 
roslaunch uuv_control_utils send_waypoints_file.launch uuv_name:=rexrov

received
 gal@gal-XPS-15-9560:~$ roslaunch uuv_control_utils send_waypoints_file.launch uuv_name:=rexrov
... logging to /home/gal/.ros/log/0c17bfdc-842c-11e8-be38-9cb6d0f2f06b/roslaunch-gal-XPS-15-9560-4543.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gal-XPS-15-9560:44615/

SUMMARY
========

PARAMETERS
 * /rexrov/send_waypoint_file/filename: /home/gal/catkin_...
 * /rexrov/send_waypoint_file/interpolator: lipb
 * /rexrov/send_waypoint_file/start_time: 0
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /rexrov/
    send_waypoint_file (uuv_control_utils/send_waypoint_file.py)

ROS_MASTER_URI=http://localhost:11311

process[rexrov/send_waypoint_file-1]: started with pid [4560]
[INFO] [1531220231.544487, 0.000000]: Send a waypoint file, namespace=/rexrov/
Traceback (most recent call last):
  File "/home/gal/catkin_ws/src/uuv_simulator/uuv_control/uuv_control_utils/scripts/send_waypoint_file.py", line 68, in <module>
    String(interpolator))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 727, in receive_once
    p.read_messages(b, msg_queue, sock) 
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
rospy.service.ServiceException: service [/rexrov/init_waypoints_from_file] responded with an error: error processing request: field success is not a bool
[rexrov/send_waypoint_file-1] process has died [pid 4560, exit code 1, cmd /home/gal/catkin_ws/src/uuv_simulator/uuv_control/uuv_control_utils/scripts/send_waypoint_file.py __name:=send_waypoint_file __log:=/home/gal/.ros/log/0c17bfdc-842c-11e8-be38-9cb6d0f2f06b/rexrov-send_waypoint_file-1.log].
log file: /home/gal/.ros/log/0c17bfdc-842c-11e8-be38-9cb6d0f2f06b/rexrov-send_waypoint_file-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Any suggested ?

Many thanks

Gal

musamarcusso, 2018-07-10:
Hi @galfunk, could you send the output from the controller node? I tried reproducing it here, but I didn't have the same issue.

galfunk, 2018-07-10:
Hi I'm not sure what you asking for... I'm pretty new in the simulation environment.
Where I can find the  output from the controller node ?

musamarcusso, 2018-07-10:
Did you call this launch file after starting, for example, the DP controller tutorial demo? If that is so, it can be that the error occurred in the callback function for this service.

galfunk, 2018-07-10:
I call this launch file after used 
roslaunch uuv_tutorial_dp_controller start_tutorial_dp_controller_demo.launch

galfunk, 2018-07-10:
Now I try to call first 
roslaunch uuv_control_utils send_waypoints_file.launch uuv_name:=rexrov

and receive
gal@gal-XPS-15-9560:~$ roslaunch uuv_control_utils send_waypoints_file.launch uuv_name:=rexrov
... logging to /home/gal/.ros/log/64a37ebe-845b-11e8-be38-9cb6d0f2f06b/roslaunch-gal-XPS-15-9560-18187.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gal-XPS-15-9560:41449/

SUMMARY
========

PARAMETERS
 * /rexrov/send_waypoint_file/filename: /home/gal/catkin_...
 * /rexrov/send_waypoint_file/interpolator: lipb
 * /rexrov/send_waypoint_file/start_time: 0
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /rexrov/
    send_waypoint_file (uuv_control_utils/send_waypoint_file.py)

auto-starting new master
process[master]: started with pid [18197]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 64a37ebe-845b-11e8-be38-9cb6d0f2f06b
process[rosout-1]: started with pid [18210]
started core service [/rosout]
process[rexrov/send_waypoint_file-2]: started with pid [18219]
[INFO] [1531238870.711556]: Send a waypoint file, namespace=/rexrov/
Traceback (most recent call last):
  File "/home/gal/catkin_ws/src/uuv_simulator/uuv_control/uuv_control_utils/scripts/send_waypoint_file.py", line 56, in <module>
    raise rospy.ROSException('Service not available! Closing node...')
rospy.exceptions.ROSException: Service not available! Closing node...
[rexrov/send_waypoint_file-2] process has died [pid 18219, exit code 1, cmd /home/gal/catkin_ws/src/uuv_simulator/uuv_control/uuv_control_utils/scripts/send_waypoint_file.py __name:=send_waypoint_file __log:=/home/gal/.ros/log/64a37ebe-845b-11e8-be38-9cb6d0f2f06b/rexrov-send_waypoint_file-2.log].
log file: /home/gal/.ros/log/64a37ebe-845b-11e8-be38-9cb6d0f2f06b/rexrov-send_waypoint_file-2*.log

Thanks again for your assistance, I really appreciated

Gal

musamarcusso, 2018-07-10:
In this last case it will not work because the controller node has not been started. The service to receive the waypoint file won't be available. When you start the tutorial demo and then the `send_waypoint_file.py` script, what is the output from the tutorial demo? Are you using the waypoint file available as an example or did you add your own?

galfunk, 2018-07-11:
I'm using the waypoint file as an example. The output from the start_tutorial_dp_controller_demo.launch

gal@gal-XPS-15-9560:~$ roslaunch uuv_tutorial_dp_controller start_tutorial_dp_controller_demo.launch
... logging to /home/gal/.ros/log/8993f558-84df-11e8-b770-9cb6d0f2f06b/roslaunch-gal-XPS-15-9560-3548.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gal-XPS-15-9560:44229/

SUMMARY
========

PARAMETERS
 * /publish_world_models/meshes/east/plane: [2000, 0.1, 100]
 * /publish_world_models/meshes/east/pose/position: [0, 1000, -50]
 * /publish_world_models/meshes/heightmap/mesh: package://uuv_des...
 * /publish_world_models/meshes/heightmap/model: sand_heightmap
 * /publish_world_models/meshes/north/plane: [0.1, 2000, 100]
 * /publish_world_models/meshes/north/pose/position: [1000, 0, -50]
 * /publish_world_models/meshes/seafloor/plane: [2000, 2000, 0.1]
 * /publish_world_models/meshes/seafloor/pose/position: [0, 0, -100]
 * /publish_world_models/meshes/south/plane: [0.1, 2000, 100]
 * /publish_world_models/meshes/south/pose/position: [-1000, 0, -50]
 * /publish_world_models/meshes/west/plane: [2000, 0.1, 100]
 * /publish_world_models/meshes/west/pose/position: [0, -1000, -50]
 * /rexrov/ground_truth_to_tf_rexrov/child_frame_id: /rexrov/base_link
 * /rexrov/ground_truth_to_tf_rexrov/footprint_frame_id: /rexrov/base_foot...
 * /rexrov/ground_truth_to_tf_rexrov/frame_id: world
 * /rexrov/ground_truth_to_tf_rexrov/odometry_topic: /rexrov/pose_gt
 * /rexrov/ground_truth_to_tf_rexrov/stabilized_frame_id: /rexrov/base_stab...
 * /rexrov/robot_description: <?xml version="1....
 * /rexrov/robot_state_publisher/publish_frequency: 5
 * /rexrov/robot_state_publisher/robot_description: /rexrov/robot_des...
 * /rexrov/thruster_allocator/tam: [[0.1597426884844...
 * /rexrov/thruster_manager/base_link: base_link
 * /rexrov/thruster_manager/conversion_fcn: proportional
 * /rexrov/thruster_manager/conversion_fcn_params/gain: 0.00031
 * /rexrov/thruster_manager/max_thrust: 2000.0
 * /rexrov/thruster_manager/tf_prefix: rexrov
 * /rexrov/thruster_manager/thruster_frame_base: thruster_
 * /rexrov/thruster_manager/thruster_topic_prefix: thrusters/
 * /rexrov/thruster_manager/thruster_topic_suffix: /input
 * /rexrov/thruster_manager/timeout: -1
 * /rexrov/thruster_manager/update_rate: 50
 * /rexrov/tutorial_dp_controller/Kd: [9077.459, 9077.4...
 * /rexrov/tutorial_dp_controller/Ki: [321.417, 321.417...
 * /rexrov/tutorial_dp_controller/Kp: [11993.888, 11993...
 * /rexrov/tutorial_dp_controller/saturation: 5000
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /use_sim_time: True

NODES
  /rexrov/
    ground_truth_to_tf_rexrov (message_to_tf/message_to_tf)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    thruster_allocator (uuv_thruster_manager/thruster_allocator.py)
    trajectory_marker_publisher (uuv_control_utils/trajectory_marker_publisher.py)
    tutorial_dp_controller (uuv_tutorial_dp_controller/tutorial_dp_controller.py)
    urdf_spawner (uuv_descriptions/spawn_model.py)
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    publish_world_models (uuv_assistants/publish_world_models.py)
    rviz (rviz/rviz)
    world_ned_frame_publisher (tf2_ros/static_transform_publisher)

auto-starting new master
process[master]: started with pid [3562]
[... 290 lines cut ...]
[ INFO] [1531295629.030874510, 0.114000000]: Physics dynamic reconfigure ready.
[ INFO] [1531295629.050817020, 0.134000000]: Physics dynamic reconfigure ready.
Transform world_ned (NED) to world (ENU)=
[[ -3.74939946e-33   1.00000000e+00   1.22464680e-16]
 [  1.00000000e+00  -4.44089210e-16  -2.46519033e-32]
 [  0.00000000e+00   1.22464680e-16  -1.00000000e+00]]
REXROV -- 2018-07-11 08:53:49,081 | INFO | dp_controller_base | DP controller successfully initialized
Kd=
[[  9077.459      0.         0.         0.         0.         0.   ]
 [     0.      9077.459      0.         0.         0.         0.   ]
 [     0.         0.      9077.459      0.         0.         0.   ]
 [     0.         0.         0.     18880.925      0.         0.   ]
 [     0.         0.         0.         0.     18880.925      0.   ]
 [     0.         0.         0.         0.         0.     18880.925]]
Ki=
[[  321.417     0.        0.        0.        0.        0.   ]
 [    0.      321.417     0.        0.        0.        0.   ]
 [    0.        0.      321.417     0.        0.        0.   ]
 [    0.        0.        0.     2096.951     0.        0.   ]
 [    0.        0.        0.        0.     2096.951     0.   ]
 [    0.        0.        0.        0.        0.     2096.951]]
[Wrn] [msgs.cc:1808] Conversion of sensor type[gps] not suppported.
[Wrn] [msgs.cc:1808] Conversion of sensor type[gps] not suppported.
REXROV -- 2018-07-11 08:53:49,123 | INFO | wp_trajectory_generator | Interpolation method set: lipb
REXROV -- 2018-07-11 08:53:49,123 | INFO | dp_controller_local_planner | AUTOMATIC MODE = OFF

musamarcusso, 2018-07-11:
Did you call `send_waypoint_file.py` in this last case? I can't find the log from the callback function.

galfunk, 2018-07-14:
I'm re-install the UUV simulator and the problem solved.

Thanks

closed 2018-07-14 by galfunk