uuvsimulator/uuv_simulator #275: Gazebo Segfaults when I try to add a new model to a world

dbiegl, 2018-07-10:
Im trying to add a new model to an existing world. I used Munkholmen and the Herkules as examples to make a model pipe that I added to world_models. From there I added code to the world launch file to spawn the model pipe. Heres the output when I try to launch the world. 

daniel@daniel-Precision-T1650:~/catkin_ws$ roslaunch uuv_descriptions munkholmen.launch
... logging to /home/daniel/.ros/log/4fba81e0-845c-11e8-aea5-b8ca3a8b9ea8/roslaunch-daniel-Precision-T1650-9670.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://daniel-Precision-T1650:37485/

SUMMARY
========

PARAMETERS
 * /publish_world_models/meshes/herkules_ship_wreck/mesh: package://uuv_des...
 * /publish_world_models/meshes/herkules_ship_wreck/pose/orientation: [0, 0, 5.24]
 * /publish_world_models/meshes/herkules_ship_wreck/pose/position: [1052.025, 158.03...
 * /publish_world_models/meshes/island/mesh: package://uuv_des...
 * /publish_world_models/meshes/island/pose/position: [-103.391, -121.4...
 * /publish_world_models/meshes/seabed/mesh: package://uuv_des...
 * /publish_world_models/meshes/seabed/pose/position: [0, 0, 0]
 * /publish_world_models/meshes/undersea_pipe/mesh: package://uuv_des...
 * /publish_world_models/meshes/undersea_pipe/pose/orientation: [0, 0, 0]
 * /publish_world_models/meshes/undersea_pipe/pose/position: [200, 156, -40]
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    publish_world_models (uuv_assistants/publish_world_models.py)
    world_ned_frame_publisher (tf2_ros/static_transform_publisher)

auto-starting new master
process[master]: started with pid [9680]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4fba81e0-845c-11e8-aea5-b8ca3a8b9ea8
process[rosout-1]: started with pid [9693]
started core service [/rosout]
process[gazebo-2]: started with pid [9709]
process[world_ned_frame_publisher-3]: started with pid [9722]
process[publish_world_models-4]: started with pid [9723]
Gazebo multi-robot simulator, version 7.0.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[ INFO] [1531239265.241971133]: Finished loading Gazebo ROS API Plugin.
[Msg] Waiting for master.
[ INFO] [1531239265.242521177]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 129.59.105.140
Start publishing vehicle footprints to RViz
Segmentation fault (core dumped)
[gazebo-2] process has died [pid 9709, exit code 139, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver --verbose -e ode worlds/munkholmen.world __name:=gazebo __log:=/home/daniel/.ros/log/4fba81e0-845c-11e8-aea5-b8ca3a8b9ea8/gazebo-2.log].
log file: /home/daniel/.ros/log/4fba81e0-845c-11e8-aea5-b8ca3a8b9ea8/gazebo-2*.log
^C[world_ned_frame_publisher-3] killing on exit
[publish_world_models-4] killing on exit
/gazebo/get_model_state service is unavailable
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Im also running gazebo headless as my system doesn't have a dedicated graphics driver. 
Thanks for any help.

musamarcusso, 2018-07-10:
Hi @dbiegl, 
I can't really figure out the problem from this output. Just to help me understand the issue, 
1. Did you had the pipe Gazebo model to some catkin package in your workspace (such as the models in the [`world_models`](https://github.com/uuvsimulator/uuv_simulator/tree/master/uuv_descriptions/world_models) folder?
2. In case you did, if you start Gazebo using `roslaunch gazebo_ros empty_world.launch`, for example,   can you add the model from the list shown in the `Insert` tab?

dbiegl, 2018-07-10:
I based my model package off of the Herkules package. When I open an empty world I see my model in the insert tab, however I cannot actually insert it into the world. 
Heres the model.sdf
`<?xml version="1.0"?>
<sdf version="1.5">
  <model name="pipe">
    <static>true</static>
    <link name="link">
      <collision name="pipe">
      <pose>0 0 0 0 0 0</pose>
        <geometry>
          <mesh>
            <uri>model://undersea_pipe/meshes/pipe.dae</uri>
            <scale>1.0 1.0 1.0</scale>
          </mesh>
        </geometry>
      </collision>

      <visual name="pipe">
        <cast_shadows>true</cast_shadows>
        <pose>0 0 0 0 0 0</pose>
        <geometry>
          <mesh>
            <uri>model://undersea_pipe/meshes/pipe.dae</uri>
            <scale>1.0 1.0 1.0</scale>
          </mesh>
        </geometry>
      </visual>
    </link>
  </model>
</sdf>`

heres the model.config
`<?xml version="1.0"?>
<model>
  <name>undersea_pipe</name>
  <version>0.0.1</version>
  <author>
    <name>Dani Biegl</name>
    <email>dbiegl@purdue.edu</email>
  </author>
  <description>
    A simple straight pipe used for simulations
  </description>
</model>`

thanks

dbiegl, 2018-07-10:
Well that didn't work too well, I added all the files to a repository 
[https://github.com/dbiegl/Simple-Pipe](url)

musamarcusso, 2018-07-10:
One thing that is very important to make your Gazebo model visible is to add your model's path to Gazebo's models path. If, for example, you create a catkin package named `pipe_description` (this is just an example, you can name the package whatever you want) as follows

```
cd ~/catkin_ws/src
catkin create pkg pipe_descripton
```

you can then create a folder `models` in `pipe_description` and copy your `Single-Pipe` folder into `models`. Considering that you already have model's description files in `Single-Pipe`, namely `model.config`, `model.sdf` and evetually some meshes and textures (I couldn't open your link, so I could not check the files), the next step is to configure your catkin package so that your model's path is visible to Gazebo.
That means that you have to open the `pipe_description/package.xml` file and add your `pipe_description/models` path to Gazebo's model path:

```xml
<?xml version="1.0"?>
<package format="2">
  <name>pipe_description</name>
  <version>0.0.0</version>
  <description>The pipe_description package</description>

  <maintainer email=""></maintainer>

  <license>TODO</license>

  <buildtool_depend>catkin</buildtool_depend>

  <exec_depend>gazebo_ros</exec_depend>

  <export>
    <gazebo_ros gazebo_model_path="${prefix}/models"/>
  </export>
</package>
```

In this case, after you compile the package (`caktin build`) and source it again (`source ~/catkin_ws/devel/setup.bash`), you should be able to start Gazebo with the command `roslaunch gazebo_ros empty_world.launch` and see your model listed in the `Insert` tab along with other Gazebo models.

Can you check if your package is configure this way?

musamarcusso, 2018-07-10:
I just accessed your model from your profile's page. Apart from my suggestions above, I would also check the path to your `pipe.dae` mesh [here](https://github.com/dbiegl/Simple-Pipe/blob/47d59d800e61c8d8d3fb35d7c35bfef62e857e95/model.sdf#L10). According to the path, you should also create a `meshes` folder in the `Single-Pipe` folder and move `pipe.dae` to this `meshes` folder.

dbiegl, 2018-07-10:
I have the pipe.dae in a meshes folder, i forgot to copy it over when i submitted the reprository. I'll work on adding the folder to gazebo's path

musamarcusso, 2018-07-10:
Also, I tried opening the `pipe.dae` file in MeshLab, but I got an error 

```
LOG: 0 Warning: Mesh pipe.dae has not been opened
```

Maybe something happened when exporting the mesh to COLLADA.

dbiegl, 2018-07-10:
Ill take another look at the pipe.dae file again as I may have messed something up when making the mesh.

musamarcusso, 2018-07-10:
Are you exporting it in Blender?

dbiegl, 2018-07-10:
yes

musamarcusso, 2018-07-10:
The Blender COLLADA exporter is not quite verbose, you can try importing the same COLLADA file again in Blender first to check if the meshes and textures are still there. You might also try to export it as OBJ (includes textures) or STL (no textures).

dbiegl, 2018-07-10:
alright ill play around with blender some, if that doesn't work do you have a recommended software?

musamarcusso, 2018-07-10:
I think you can convert meshes using MeshLab. Only if you need to edit the meshes, Blender might be the only free option to use, as far as I know.

dbiegl, 2018-07-11:
I finally fixed this issue and everything is running smoothly now. I remade the files and made sure names matched across folders and I added an stl. Thanks

closed 2018-07-11 by dbiegl