uuvsimulator/uuv_simulator #286: Tests not being run due to syntax errors in CMakeLists.txt

g-arjones, 2018-07-19:
There is a syntax error in uuv_gazebo_ros_plugins' and uuv_manipulators_kinematics' CMakeLists:

In uuv_gazebo_ros_plugins/CMakeLists.txt:

```
  foreach(UNIT_TESTS
    test/test_default_fossen_vehicle.py
    test/test_sphere_vehicle.py
    test/test_thrusters.py
    catkin_add_nosetests(${UNIT_TESTS}))
  endforeach()
```

The closing ')' should be after the list of python files (ending foreach's argument list) and in uuv_manipulators_kinematics/CMakeLists.txt there is a similar syntax error.

I would fix this and open a PR but unfortunately the unit tests for these packages are either failing or broken..

musamarcusso, 2018-07-20:
Thanks for catching this.
The tests run if I run them individually

```
mam@pc:~/catkin_ws$ rosrun uuv_gazebo_ros_plugins test_default_fossen_vehicle.py 
[ROSUNIT] Outputting test results to /home/mam/.ros/test_results/uuv_gazebo_ros_plugins/rosunit-test_default_fossen_vehicle.xml
[Testcase: test_added_mass_offset] ... ok
[Testcase: test_added_mass_scaling] ... ok
[Testcase: test_damping_scaling] ... ok
[Testcase: test_get_model_parameters] ... ok
[Testcase: test_linear_damping_offset] ... ok
[Testcase: test_linear_forward_speed_damping_offset] ... ok
[Testcase: test_set_fluid_density] ... ok
[Testcase: test_volume_offset] ... ok
[Testcase: test_volume_scaling] ... ok
-------------------------------------------------------------
SUMMARY:
 * RESULT: SUCCESS
 * TESTS: 9
 * ERRORS: 0 []
 * FAILURES: 0 []

mam@pc:~/catkin_ws$ rosrun uuv_gazebo_ros_plugins test_sphere_vehicle.py 
[ROSUNIT] Outputting test results to /home/mam/.ros/test_results/uuv_gazebo_ros_plugins/rosunit-test_sphere_vehicle.xml
[Testcase: test_added_mass_coefs] ... ok
[Testcase: test_get_model_parameters] ... ok
[Testcase: test_nonlinear_damping_coefs] ... ok
-------------------------------------------------------------
SUMMARY:
 * RESULT: SUCCESS
 * TESTS: 3
 * ERRORS: 0 []
 * FAILURES: 0 []

mam@pc:~/catkin_ws$ rosrun uuv_gazebo_ros_plugins test_thrusters.py 
[ROSUNIT] Outputting test results to /home/mam/.ros/test_results/uuv_gazebo_ros_plugins/rosunit-test_thrusters.xml
[Testcase: test_change_dyn_state_efficiency] ... ok
[Testcase: test_change_thrust_efficiency] ... ok
[Testcase: test_change_thruster_state] ... ok
[Testcase: test_convertion_fcn_parameters] ... ok
[Testcase: test_input_output_topics_exist] ... ok
-------------------------------------------------------------
SUMMARY:
 * RESULT: SUCCESS
 * TESTS: 5
 * ERRORS: 0 []
 * FAILURES: 0 []
```

I will check this error.

musamarcusso, 2018-07-20:
Now it seems to be working. The [Travis build](https://travis-ci.org/uuvsimulator/uuv_simulator/jobs/406148076#L4201) ran the tests as expected.

g-arjones, 2018-07-20:
Great... Thanks.

What about [these](https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_manipulators/uuv_manipulators_commons/uuv_manipulators_kinematics/CMakeLists.txt#L39)?

musamarcusso, 2018-07-20:
I am also fixing those, but the whole manipulators package is supposed to be moved somewhere else soon.

g-arjones, 2018-07-20:
Awesome, Musa... Thanks a lot!

closed 2018-07-23 by g-arjones