uuvsimulator/uuv_simulator #287: Protobuf incompatibility errors in Ubuntu 16.04 with ROS Kinetic, Gazebo 7

AkshayHinduja, 2018-07-19:
I am trying to compile the package on my computer but running into errors similar to:-

```
/usr/include/gazebo-7/gazebo/msgs/world_control.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/usr/include/gazebo-7/gazebo/msgs/world_control.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
```
I was initially using Protobuf 2.6.1 but seeing the errors migrated to 3.3.0 by manually upgrading the version. It didn't make any difference to the errors. I have also reinstalled gazebo but to no avail. The compilation worked on a different system with the same configuration, with protobuf 2.6.1 
Any suggestions on how to resolve this?

musamarcusso, 2018-07-20:
Hi @AkshayHinduja, just to clarify, does this happen when running `catkin build` in your workspace? I never run into this issues when using [Gazebo's installation instructions](http://gazebosim.org/tutorials?tut=install_ubuntu&cat=install)

AkshayHinduja, 2018-07-20:
Hi @musamarcusso. Yes it happens while running `catkin build` in the workspace. 
To clarify, this issue is not during Gazebo's installation, it is when building the uuv_simulator. 
I removed gazebo-7 (pre-installed with ros-kinetic) to freshly install gazebo 9, but the same problem persists. 

Another example of the error messages:-

`In file included from /usr/include/ignition/msgs1/ignition/msgs/MessageTypes.hh:163:0,
                 from /usr/include/ignition/msgs1/ignition/msgs/Utility.hh:29,
                 from /usr/include/ignition/msgs1/ignition/msgs.hh:25,
                 from /usr/include/ignition/transport4/ignition/transport/Node.hh:32,
                 from /usr/include/gazebo-9/gazebo/physics/Entity.hh:25,
                 from /usr/include/gazebo-9/gazebo/physics/Link.hh:37,
                 from /home/akshay/hauv_sim_ws/src/uuv_simulator/uuv_world_plugins/uuv_world_plugins/src/UnderwaterCurrentPlugin.cc:24:
/usr/include/ignition/msgs1/ignition/msgs/joint_cmd.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/usr/include/ignition/msgs1/ignition/msgs/joint_cmd.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/usr/include/ignition/msgs1/ignition/msgs/joint_cmd.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
`

musamarcusso, 2018-07-23:
Have you tried reinstalling only the `ignition` libraries? The problems seems to be around `ignition`.

AkshayHinduja, 2018-07-24:
the protobuf binaries didn't work for me. It seems the issue arises from gazebo picking up a different version of ignition(from earlier binary installs) which was dependent on an older version of protoc.  (see here: http://answers.gazebosim.org/question/17820/compilation-fails-as-gazebo-and-ignition-msgs-require-different-protobuf-versions/) 
I am trying to build it from source and will see if it works. If not, then I think a fresh install is the only option. 
Thanks!

musamarcusso, 2018-07-24:
What version of Gazebo are you using?

AkshayHinduja, 2018-07-24:
Initially had Gazebo7 from the ros-kinetic install. That didnt work, tried Gazebo-9 from binaries. Now trying gazebo-9 from source

musamarcusso, 2018-07-24:
That is very strange. I don't know many other tips that I could give in that case. 
One of the things you could maybe check (I don't know how it would help) is to check the configuration for the [Travis build](https://github.com/uuvsimulator/uuv_simulator/blob/master/.uuv_ci_config/install_gazebo9.sh) since one of the cases installs Gazebo 9 for tests. But it is not much different from what the official Gazebo installation instructions might provide.

AkshayHinduja, 2018-07-24:
I finally got it compiled. Gazebo, ignition, and protobuf have a lot of compatibility issues between versions. I'll list down what I did so it might be useful to others. 

What worked for me was to first erase all gazebo, ignition and protobuf related files and binaries (sudo apt-get remove, and delete /usr/ and /usr/local include files. 

Then install from source protobuf 2.6.1. Install protobuf-compiler and protobuf-c-compiler from binaries (both are version 2.6.1 in ubunutu 16.04) 

Next is to install gazebo9 from source. (http://gazebosim.org/tutorials?tut=install_from_source&cat=install) The dependencies mentioned in their page were strangely incompatible. Best is to install the binaries which the `cmake ../` output tells for gazebo. 

After that install the ros dependencies for gazebo9. `sudo apt-get install ros-kinetic-gazebo9-ros-pkgs ros-kinetic-gazebo9-ros-control`. 

Once that is done everything should work out fine. 
The issue probably arises from newer versions of protobuf being used earlier.

JKTesla, 2019-11-27:
> I finally got it compiled. Gazebo, ignition, and protobuf have a lot of compatibility issues between versions. I'll list down what I did so it might be useful to others.
>
> What worked for me was to [...]

Hi @AkshayHinduja , thanks for your advice. I also do not know when my ubuntu 16.04's protobuf version changed from 2.6.1 to 3.3.0. 

I accepted your advice to install gazebo9 from source, **but I wanted to maintain protobuf version in 3.3.0** . 

I find the core of problem is ignition (the gazebo required dependencies) binaries. So I try to install ignition from source and compile them ([SDFormat…](http://gazebosim.org/tutorials?tut=install_dependencies_from_source) 、[Ignition Transport](https://ignitionrobotics.org/tutorials/transport/4.0/md__data_ignition_ign-transport_tutorials_installation.html)、[DART](http://dartsim.github.io/install_dart_on_ubuntu.html#install-dart-from-source)) using the default protobuf 3.3.0.  The uuv_simulator is able to compile and work well using gazebo-9 in protobuf 3.3.0.

**Note:** The default branch your cloned repository about ignition and gazebo are the development branch, you should use `hg up or git check` to switch to the branch of the corresponding version (e.g. sdf6 for Gazebo 9) .