uuvsimulator/uuv_simulator #291: Releasing uuv_simulator to build farm

Nirzvi1, 2018-07-23:
Hi,

My organization (Clearpath Robotics) is using uuv_simulator to simulate the Heron's water physics. To simplify the process for Heron customers, it would be great if this package could be released so they wouldn't have to install from source.

Thanks

Nirzvi1, 2018-07-27:
The rotors_simulator method seems like it should work fine for our purposes, so I vote for that method. I also like the extra simplicity.
Thanks for doing this!

tonybaltovski, 2018-08-02:
@musamarcusso I think this might need to have distro specific branches since Gazebo changes version depending on the ros distro.  Having some experience maintaining, I generally have a branch for the ros version the repo was developed for.  I continue to release that in new ros distros till there are changes specific to that distro or I want to implement breaking changes/behaviours.  With that said, you change which branch is being used to do the release later.

musamarcusso, 2018-08-03:
Hi @tonybaltovski, 
that is a good point, the changes in Gazebo have to be considered in these cases. I don't know how easy or sustainable it can be to keep the current method (the plugins not have tests for the current `GAZEBO_MAJOR_VERSION` to have code adapted to each version of the Gazebo API).
I am reading currently the thread on ROS Discourse on [branching strategies for ROS projects](https://discourse.ros.org/t/on-git-branching-strategies/4785) to have some more information.

musamarcusso, 2018-08-14:
Just an update: I am waiting for the needed authorization on my end here to make this release. I hope it will be granted as soon as possible.

tonybaltovski, 2018-08-14:
@musamarcusso awesome, thanks for the update!

jellevos, 2019-03-27:
@musamarcusso Is there any way we can do this manually with the current version? UUV simulator is currently in this list: https://github.com/ros/rosdistro/blob/master/melodic/distribution.yaml

But trying to actually depend on it results in an error during rosdep install.

gijskoning, 2019-04-06:
I just checked and the packages are released and available to install!

pxalcantara, 2020-05-19:
@musamarcusso  and @guystoppi I think we can close this issue, right?!