uuvsimulator/uuv_simulator #293: Include ROSBaseModelPlugin

arturmiller, 2018-07-24:
Hello,

I am currently writing a custom gazebo sensor plugin. For that, I want my plugin to inherit from ROSBaseModelPlugin from the uuv_sensor_ros_plugins package. Unfortunately that isn't possible. I get the error message:
"[Plugin.hh:165] Failed to load plugin {myplugin}: {mypluginpath}: undefined symbol: _ZTIN6gazebo18ROSBaseModelPluginE"
The problem is that the uuv_sensor_ros_plugins package doesn't export the library. This could be solved by adapting the CMakeLists.txt file.
(ROSBaseSensorPlugin should also be exported)

Best regards

musamarcusso, 2018-07-24:
I just merged a patch to this issue. Could you give a short feedback if this solves the problem?

arturmiller, 2018-07-25:
Your patch has solved this issue.
Thank you  very much!

closed 2018-07-25 by arturmiller