uuvsimulator/uuv_simulator #296: Problems with gzserver. Assertion. [gazebo-2] exit code 134

juanscelyg, 2018-07-26:
Hi everyone,

I have a problem when I launch the rov launch to spawn into the world. The rov has made by myself. However, this problem doesnt pass when I launch the RexRov launch. It is so complicated to show the every files in the project but if you need to see the problem, you just tell me and I will load the files.

The error is below:

`gzserver: /home/juanscelyg/catkin_ws/src/uuv_simulator/uuv_gazebo_plugins/uuv_gazebo_plugins/src/UnderwaterObjectPlugin.cc:210: virtual void gazebo::UnderwaterObjectPlugin::Update(const gazebo::common::UpdateInfo&): Assertion `(!std::isnan(linearAccel) && !std::isnan(angularAccel))&&("Linear or angular accelerations are invalid.")' failed.
Aborted (core dumped)
[gazebo-2] process has died [pid 30242, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver --verbose -e ode worlds/empty_underwater.world __name:=gazebo __log:=/home/juanscelyg/.ros/log/7207b908-9103-11e8-b86e-90e6ba9d2a9a/gazebo-2.log].
log file: /home/juanscelyg/.ros/log/7207b908-9103-11e8-b86e-90e6ba9d2a9a/gazebo-2*.log
`

Thanks a lot!!

juanscelyg, 2018-07-27:
Finally, I found the error. 

I had the drag parameters with positive values, the simulator needs always negative values. It was a stupid error, I know it.

Close the issue.

closed 2018-07-27 by juanscelyg

Ankitpatel141, 2021-11-25:
Hi, I am getting similar error. Could you please elaborate your solution. I am new to ros(a month). I have tried to launch gazebo with launch file and got this error. I don't know if it's a problem with my system or launch file. Could you please help me on this.

Error:
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [62507]
[INFO] [gzclient   -2]: process started with pid [62509]
[gzserver-1] [INFO] [1637856684.181392578] [dolly.diff_drive]: Wheel pair 1 separation set to [1.250000m]
[gzserver-1] [INFO] [1637856684.181698023] [dolly.diff_drive]: Wheel pair 1 diameter set to [0.300000m]
[gzserver-1] [INFO] [1637856684.182492372] [dolly.diff_drive]: Subscribed to [/dolly/cmd_vel]
[gzserver-1] [INFO] [1637856684.183361463] [dolly.diff_drive]: Advertise odometry on [/dolly/odom]
[gzserver-1] [INFO] [1637856684.184274027] [dolly.diff_drive]: Publishing odom transforms between [odom_demo] and [chassis]
[gzclient   -2] gzclient: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::rendering::Camera; typename boost::detail::sp_member_access<T>::type = gazebo::rendering::Camera*]: Assertion `px != 0' failed.
[gzclient   -2] Aborted (core dumped)
[ERROR] [gzclient   -2]: process has died [pid 62509, exit code 134, cmd 'gzclient   '].

ros2 foxy ubuntu 20
Intel® Core™ i7-8565U CPU @ 1.80GHz × 8 
Mesa Intel® UHD Graphics 620 (WHL GT2)