uuvsimulator/uuv_simulator #308: Is absolute/gps position necessary?

szat, 2018-09-24:
Dear all,

I have just started looking at this amazing project. I saw that the topic
/rexrov/gps
seemed unused with no subscribers, but on the other hand the node "VelocityControl" was subscribing to odometry from the topic
/rexrov/post_gt
But unwater robots don't have access to gps and hence no access to absolute position, right? How does that work, what are the poses in /rexrov/post_gt that are being used? 
If someone could let me know I would greatly appreciate!

Sincerely,
Adrian

szat, 2018-09-24:
I have the feeling that it has to do with this bit of code:
```
    def odometry_callback(self, msg):
        """Handle updated measured velocity callback."""
        if not bool(self.config):
            return

        linear = msg.twist.twist.linear
        angular = msg.twist.twist.angular
        v_linear = numpy.array([linear.x, linear.y, linear.z])
        v_angular = numpy.array([angular.x, angular.y, angular.z])

        if self.config['odom_vel_in_world']:
            # This is a temp. workaround for gazebo's pos3d plugin not behaving properly:
            # Twist should be provided wrt child_frame, gazebo provides it wrt world frame
            # see http://docs.ros.org/api/nav_msgs/html/msg/Odometry.html
            xyzw_array = lambda o: numpy.array([o.x, o.y, o.z, o.w])
            q_wb = xyzw_array(msg.pose.pose.orientation)
            R_bw = trans.quaternion_matrix(q_wb)[0:3, 0:3].transpose()

            v_linear = R_bw.dot(v_linear)
            v_angular = R_bw.dot(v_angular)

        # Compute compute control output:
        t = msg.header.stamp.to_sec()
        e_v_linear = (self.v_linear_des - v_linear)
        e_v_angular = (self.v_angular_des - v_angular)

        a_linear = self.pid_linear.regulate(e_v_linear, t)
        a_angular = self.pid_angular.regulate(e_v_angular, t)

        # Convert and publish accel. command:
        cmd_accel = geometry_msgs.Accel()
        cmd_accel.linear = geometry_msgs.Vector3(*a_linear)
        cmd_accel.angular = geometry_msgs.Vector3(*a_angular)
        self.pub_cmd_accel.publish(cmd_accel)
```
Is that correct, do I need to change something here in order to not use global / gps coordinates?

pxalcantara, 2020-05-19:
Hi @szat is this issue still makes sense?