uuvsimulator/uuv_simulator #324: Wrong package name in quick start in docs

BAJUKA, 2019-01-06:
In the quick start of the docs, it is given,
roslaunch uuv_descriptions empty_underwater_world.launch
but I guess it should be
roslaunch uuv_gazebo_worlds empty_underwater_world.launch
since this launch file is not present in uuv_descriptions but present in uuv_gazebo_worlds

milan-r-shah, 2019-03-31:
Hi Bajuka,

I'm also facing the same issue but after following your instructions I found that both commands:
01. _**roslaunch uuv_gazebo_worlds empty_underwater_world.launch**_
02. _**roslaunch uuv_gazebo_worlds ocean_waves.launch**_
are giving the same output.
Is it a bug and you are also getting the same thing?

Thanks a lot in advance!

BAJUKA, 2019-04-01:
Running the above two commands worked for me. Did you  **_source devel/setup.bash_** in your workspace before you ran these commands?

milan-r-shah, 2019-04-02:
> Running the above two commands worked for me. Did you **_source devel/setup.bash_** in your workspace before you ran these commands?

Hi @BAJUKA ,
Thanks a lot for your quick response.
Yes, I ran **_source devel/setup.bash_** . In fact, I have put that command in my .bashrc file. The problem is for two different launch files (uuv_gazebo_worlds empty_underwater_world.launch & uuv_gazebo_worlds ocean_waves.launch), I'm getting the same output.
So, that's expected or there is any problem/bug?