uuvsimulator/uuv_simulator #332: Release for melodic

Timple, 2019-02-05:
I noticed you are in a releasing mood anyway. So maybe it is a good idea to release for ROS melodic as well. Since we plan on migrating ourselves this would save us a lot of compiling :)

musamarcusso, 2019-02-05:
Hi @Timple, I am already working on this. I have Travis building the package for `melodic` currently. I only have to figure out a dependency (`message_to_tf`) that hasn't been released for `melodic`. When I fix this I can make the release.

gijskoning, 2019-02-14:
I would really like to see this too. Is it possible to create a work around with robot localization for now?

musamarcusso, 2019-02-14:
I am looking into reimplementing or including the necessary module in this package to solve this problem. It is basically only one .cpp file so it seems to me that including it with the proper documentation into the `uuv_simulator` can be justified in order to have the release ready.

gijskoning, 2019-02-14:
I downloaded message_to_tf and it builds just fine. I tried simulating a robot and that also worked fine.

musamarcusso, 2019-02-14:
I also clone the package in the [Travis build configuration](https://github.com/uuvsimulator/uuv_simulator/blob/master/.uuv_ci_config/ros_melodic.sh) for `melodic`. It seems to be the only dependency missing

musamarcusso, 2019-02-28:
Hi @Timple, I just made the transfer of the `message_to_tf` node with proper 3rd party documentation to this repository at #345  so I think once the pre-release tests are run I can start working on the `melodic` release.

musamarcusso, 2019-03-13:
I just did the releases for `melodic` and `lunar` yesterday. Things are looking good in the Buildfarm. I think the packages should be out by the next sync.

closed 2019-03-13 by musamarcusso