uuvsimulator/uuv_simulator #340: Failed to show texture in new seabed

arainbilal, 2019-02-17:
Following the tutorial "making a simple seabed" (https://uuvsimulator.github.io/tutorials/seabed_world.html), the texture is not appearing properly. I think there is more to do in the Blender when exporting the mesh to Collada DAE file for the Gazebo world. Indeed, it can be seen in the tutorial page that the sand.material is used for a texture but when its imported in Gazebo, it is no where close to the the texture file given in sand.material. I think there are two things to be considered: 1) If the mesh grid is too small, the texture won't appear properly. The fine grid is due to the interpolation applied in the python script (see tutorial page); 2) The material scaling should be applied properly.

To this end, I have performed multiple tests in using different settings but cannot get the texture applied to the tutorial mesh (above weblink) similar to the world_models/height_sandmap. I request to please include few more steps or provide some guidance in the tutorial page or in repose to this issue about how the texture can appropriately can be applied on the tutorial seabed mesh created using Python scripts to be visualize appropriately in the Gazebo. 

I think that the answer to this issue will be helpful to all other who have less experience with Blender and are trying to create world files with appropriate texture visualizing the uuv_simulator environment.

musamarcusso, 2019-02-28:
Hi @arainbilal, thanks for the heads up. I will check the documentation. Blender is indeed quite a challenge.

arainbilal, 2019-03-01:
Hi @musamarcusso . Thanks. I have now attached the gazebo world images showing the tutorial seabed and the sand_heightmap, which may help to elaborate the issue. In both files, I have used the sames material in the sdf file as follows:
`<uri>file://tutorial_seabed_world/materials/scripts/sand.material</uri>`
The objective is to display the sand material on the tutorial seabed, similar to the one showing in attached sand_heightmap. Hope to get more info soon.

Tutorial seabed with sand.material
[image]

sand_heightmap with sand.material
[image]

closed 2022-11-25 by arainbilal