uuvsimulator/uuv_simulator #341: launch files missing for the quick start manual

szat, 2019-02-18:
Hello!
It seems that some of the launch files in uuv_simulator/uuv_descriptions/launch  are missing. 
For instance "empty_underwater_world.launch". But they seem to be there if you do 
cd catkin_ws/src/uuv_simulator
git checkout 5878f11

It would be nice to commit the descriptions in the tutorials to the latest version. 
:)

musamarcusso, 2019-02-28:
Hey @szat, thanks for seeing this, I will update everything as soon as I tackle the `kinetic` release issues. I am having now.

szat, 2019-03-01:
No worries, it's a great project you have here! I thought I'd help out a bit :)

milan-r-shah, 2019-04-01:
Hi @szat 
I'm also facing the same issue.
Thanks for drawing the attention of the contributors.