uuvsimulator/uuv_simulator #351: Incorrect Coefficient of Drag Calculation in LiftDragModel.cc

jzhu1234, 2019-03-15:
The compute function in LiftDragTwoLines::compute produces a negative drag when the angle of attack is negative. In other words, it produces thrust when the angle of attack is negative. 

This occurs because of how coefficient of drag is calculated. Unlike coefficient of lift, which should change signs based on the sign of the angle of attack (alpha), coefficient of drag should always stay positive.

It would be best to change: 
       Line 256:
       cd = this->cda * alpha; -> cd = this->cda * fabs(alpha);
and
       Line 241:
       cd = -this->cda * this->alphaStall +this->cdaStall * sumAlpha;
       -> cd = this->cda * this->alphaStall +this->cdaStall * -sumAlpha;

musamarcusso, 2019-03-15:
Hi @jzhu1234, did you manage to test this? I can try it out too, but it would be good to compare with your results.

jzhu1234, 2019-03-15:
I tested it by temporarily removing lift force so I could see the direction
of the drag forces in RViz. I used a similar model to the ECA A9, except
the fins are in a plus configuration.

With the current branch, in RViz I could see the drag forces go in opposite
directions when I told the vehicle to dive, rise, or turn.

musamarcusso, 2019-03-18:
is it possible to provide your configuration for a test scenario so I can reproduce it here?

jzhu1234, 2019-03-18:
iver_base.xacro -  The vehicle we model is smaller than the eca-a9, so the x and y values would have to be adjusted.
 
    <xacro:fin_macro namespace="${namespace}" fin_id="0">
      <origin xyz="-1.05 -0.07366 0" rpy="${90*d2r} 0 0" />
    </xacro:fin_macro>

    <xacro:fin_macro namespace="${namespace}" fin_id="1">
      <origin xyz="-1.05 0 -0.07366" rpy="${180*d2r} 0 0" />
    </xacro:fin_macro>

    <xacro:fin_macro namespace="${namespace}" fin_id="2">
      <origin xyz="-1.05 0.07366 0" rpy="${-90*d2r} 0 0" />
    </xacro:fin_macro>

    <xacro:fin_macro namespace="${namespace}" fin_id="3">
      <origin xyz="-1.05 0 0.07366" rpy="0 0 0" />
    </xacro:fin_macro>

iver_snippets.xacro
a0 = 0
astall = 0.174533
cla = 10.0
clastall = -1.0
cda = 1.55
cdastall = 1.55