uuvsimulator/uuv_simulator #359: Plugin error

Eberty, 2019-04-26:
`[Err] [Plugin.hh:165] Failed to load plugin libuuv_underwater_current_ros_plugin.so: libuuv_underwater_current_ros_plugin.so: cannot open shared object file: No such file or directory`
`[Err] [Plugin.hh:165] Failed to load plugin libuuv_sc_ros_interface_plugin.so: libuuv_sc_ros_interface_plugin.so: cannot open shared object file: No such file or directory`

closed 2019-04-26 by Eberty

Sollimann, 2019-08-27:
> If i remember the uvv updated the lib name. There was a PR in this epoch
> that they rename various libs and packages. But i don't remember the exact
> name now.

Ok, thank you. what do you mean by "There was a PR in this epoch"? Do you
remember how you managed to fix it?

Sollimann, 2019-08-27:
@Eberty I am having th exact same issue. Did you figure it out?

Sollimann, 2019-08-28:
> https://github.com/uuvsimulator/uuv_simulator/blob/aadee6a6896d4225e3a42326065e291500cdf6ab/uuv_world_plugins/uuv_world_ros_plugins/CMakeLists.txt#L44
>
> Try to install this package again, and verify if these libraries will be
> installed, if ok the files libuuv_underwater_current_ros_plugin.so and
> libuuv_sc_ros_interface_plugin.so will be created on
> CATKIN_PACKAGE_LIB_DESTINATION
>
> Maybe you will need export the plugins and load the model from the package.

Thank you soo much for your help, Eberty. Really appreciate you taking your
time to answer. So the current status now is that I have been able to
install the uuv_world_ros_plugins again, and I don't get the error that I
did when launching an underwater world. However now, when I try to upload a
vehicle, I now get the following error:

[Err] [Plugin.hh:165] Failed to load plugin
libunderwater_object_ros_plugin.so: libunderwater_object_ros_plugin.so:
cannot open shared object file: No such file or directory

libunderwater_object_ros_plugin.so seems to be a plugin connected to the
vehicle.

Any idea how to solve this? I haven't found any solution yet. I attach a
picture of the error message as well.

Eberty, 2019-08-29:
I think that is the same issue, try to find where `underwater_object_ros_plugin` and try to compile the package that create it again.