uuvsimulator/uuv_simulator #360: Camera only sees seabed when perpendicular to it

dbcesar, 2019-05-10:
Hello all, 

I'm currently having problems to see the seabed texture of the ocean_box model on my camera.

I've played around with the camera position and it seems that the camera only renders the texture when it is pointing directly to the ground. 

See these pictures:
[image]
Rotating camera to be appox. perpendicular:
[image]

Also, depending on the camera position, Gazebo GUI client only renders blue. You can see the effect on the image below.

[image]

It is probably not a uuv_simulator problem, but I want to know if you have noticed that already and whether you have any workaround for this. Since my application is a crawler, it will be all the time on the ground and would be awesome if we could see the ground texture on the image. 

Thanks!

musamarcusso, 2019-05-30:
Hi, could you send a snippet of the URDF robot description you used where you add the camera to the vehicle?

pxalcantara, 2020-05-17:
@dbcesar Any update of this problem? Are you still having it?

HashirZahir, 2020-05-24:
Hi I cant seem to replicate this on the rexrov camera. 

[image]

Rotating the vehicle in any roll or pitch does not seem to affect its view of the sea floor. Perhaps you can clarify what you meant? Also, I noticed that your AUV is below the surface sea bed in one of the screenshots, hence it cannot see the seabed surface. That is normal as the camera has to be above the surface to be able to see the surface. Maybe you can also specify exactly where the camera is in your screenshot as it is not clear. Please do clarify thanks.

dbcesar, 2020-05-25:
Hi @pxalcantara and @HashirZahir, I haven't been using the underwater simulation in my project for a while now. The current state is also that I can't easily test it. Thus, I'm going to close this issue for now and re-open it in the future in case I face this again. 

Thanks for the attention

pxalcantara, 2020-05-25:
Ok @dbcesar Let's close it them !

closed 2020-05-25 by pxalcantara